ControlSystemStudio/cs-studio #1280: Feature Request: ability to run script when an OPI is started up

b-spruck, 2015-07-30:
In contrast to #1278 I am looking for a way to run a script at opi start up. That is _before_ the macro processing is taking place.

Aim: Set macros in dependence of another macro

Simplified Example: Have a macro DEVICE=abcd set (which is used with PV names). From that I would like to extract (by some tables) the connected boards, physical/logical location (for navigation), human readable description etc.
Needless to say that we have lot of boards but want only one opi.

Bjoern

PS: One can trick that by using an empty opi and then use a macro which finally loads a linking container. But that is "slow" and not very elegant.

dmichel76, 2015-08-03:
Not sure I understand fully what you're after, but could it be solved by using "dynamic macros", i.e. using 

macroInput = DataUtil.createMacrosInput(true);
macroInput.put("my_macro", "some_value")
ScriptUtil.openOPI(widget, filePath, 0, macroInput);

?

b-spruck, 2015-08-03:
This only moves the problem to the OPI which open the new one, but does not solve it.
(if I understood correctly, you propose to run this macros after i.e. click on a button and with the macro open the new opi, right?)

dmichel76, 2015-08-03:
As I said, I'm not sure I understand what your situation is.... I guess it also depends on how you plan on launching the OPI. If it's acceptable to have a button launch that OPI, you can use a script to do so (rather than using a 'open OPI' action) and use the macroInput, as per the example code in my previous post, to dynamically set the macros you want/need.

Since you're now opening an OPI from script and can set the macros in that scripts, you can add logic there to set them to whatever it is need them to be set to

But it sounds like this doesn't address your problem - Can elaborate a bit more?

b-spruck, 2015-08-03:
O.k.
Lets say there is more than one way to open a specific opi for a 
specific board. In most cases it will be a button within another opi, 
but it could be different like clicking on an Alarm.
Now using a script anywhere seems a bit complicated and error prone, esp 
as I would have to supply the opi name to every script in addition or 
would need one script per opi.
Thus I see it a more natural and clever way to runa script right after 
loading the opi.

MichaelRitzert, 2015-08-11:
Another option might be to add a "refresh all macros" command to the widget. Presently, this doesn't work as I naively expect:

m=display.getPropertyValue('macros')
m.put('TEST2','new value')
display.setPropertyValue('macros', m)
widget.getChild('Label_1').setPropertyValue('text', '$(TEST2)')

=> the label reads "$(TEST2)", not "new value".

Maybe we could have a
display.refreshMacros()
that forces all widgets to re-evaluate.

(Ironically the "Show Macros" dialog that explicitly mentions "this won't reflect the macro changes after opening" has the new value.)

berryma4, 2015-08-11:
Yes, currently to refresh macros you need to refresh the opi.
(which is pretty much what display.refreshMacros() would need to do)
I believe there is an example in BOY Examples (probably the linking
container example does this).

b-spruck, 2015-08-11:
Michael, this would indeed solve one part of the problem, but then still you have a very visible delay and flicker if you open the opi.

MichaelRitzert, 2015-08-13:
I have some new insight: This code runs in a script in the OPI with =1 trigger:

    m=display.getPropertyValue('macros')
    m.put('TEST2','new value')
    display.setPropertyValue('macros', m)
    l1ep = display.getWidget('Label_1')
    l1ep.setPropertyValue('text', '$(TEST2)')
    #l1ep.refresh()

independent of whether I enable of disable the final refresh(), the label reads "$(TEST2)".

Now I add a push button that triggers this script:

    l1ep = display.getWidget('Label_1')
    l1ep.refresh()

i.e. exactly the same, but in a different context. And this actually updates the label text! I debugged the code and it seems the macroMap is not updated in time for the first refresh() call still within the script that changed the macros.
Can somebody explain what is happening and if there is a chance to have the map updated earlier? Then a refreshMacros action could possibly be a simple as calling refresh() on all widgets.

MichaelRitzert, 2015-08-13:
One (well, two...) more:

When I change the script in the action button to

    m=display.getPropertyValue('macros')
    m.put('TEST2','button')
    display.setPropertyValue('macros', m)

    l1ep = display.getWidget('Label_1')
    l1ep.refresh()

the first time I click the button, the label reads "new value", the second time I click it updates to "button". And:
When I put the `l1ep.refresh()` in a second script also with =1 trigger, it also doesn't update the label.

berryma4, 2015-09-29:
@dmichel76 ok to close?