ControlSystemStudio/cs-studio #1281: Alarm Table: alarm description without alarm severity prefix

utzeln, 2015-07-31:
Would it be possible to make the alarm severity prefix optional [by default: true]?

On ITER, as we intend to use the alarm icon and the alarm message background colour to reflect the alarm severity, we would like to reduce the Description column to what was configured by the user.

kasemir, 2015-07-31:
When the description text starts with "* ", the "MINOR Alarm: " prefix is omitted, so I think you can already get what you want.
See online help of alarm system 10.2 Description, which points to 11.3 Annunciation Text.

utzeln, 2015-07-31:
Right - this means that we ask our user to add * to the description of the alarm.

I thought that this was only for the annunciator.

kasemir, 2015-08-04:
Yes, description and annunciation are currently pretty much the same.

OK, so do I understand correctly that you'd like to hide the "MINOR Alarm:" from the description column in the table, since the severity is already obvious from the separate Severity column and/or icon and/or future background color, but the annunciation would remain unchanged?
Annunciation still defaults to "MINOR Alarm: whatever", unless the description starts with "* "?

I think that'll be good, we don't even need a preference.
Just change the table to always show just the description.

Technical detail: The description could be "* The water temperature reached {1}" where {1} will be replaced by the value that triggered the alarm.
I assume you'd like to show "The water temperature reached 110" and not "The water temperature reached {1}"?
So the change wouldn't just show the plain description input. It still needs to format it, but using a format that slightly differs from the one used for annunciation, omitting the prefix and the "* " handling.

utzeln, 2015-08-05:
Yes - the alarm annunciator and the alarm notifier will make use of the * prefix and by default always add the severity information, except told otherwise.

And right, the description can include {0} and {1} that have to be formatted - the alarm description is not just plain text.

Thanks.

closed 2016-04-06 by utzeln