ControlSystemStudio/cs-studio #1292: Multistate Symbol widget throwing exception for certain SVGs.

willrogers, 2015-08-05:
When moving to the `master` branch of CSS, our multistate symbol widgets began throwing `ClassCastException` for certain SVGs that were previously working:

```
2015-08-05 11:57:28.774 WARNING [Thread 1] org.csstudio.swt.widgets.symbol.SVGSymbolImage (loadDocument) - Error loading SVG image /BL14I/BL14IApp_opi/cpuZero 0.svg
java.lang.ClassCastException: org.apache.batik.dom.GenericElementNS cannot be cast to org.apache.batik.dom.svg12.BindableElement
    at org.apache.batik.bridge.svg12.BindableElementBridge.buildCompositeGraphicsNode(BindableElementBridge.java:106)
    at org.apache.batik.bridge.svg12.BindableElementBridge.createGraphicsNode(BindableElementBridge.java:86)
    at org.apache.batik.bridge.GVTBuilder.buildGraphicsNode(GVTBuilder.java:213)
    at org.apache.batik.bridge.GVTBuilder.buildComposite(GVTBuilder.java:171)
    at org.apache.batik.bridge.GVTBuilder.build(GVTBuilder.java:82)
    at org.csstudio.utility.batik.SVGHandler.<init>(SVGHandler.java:181)
```

I can supply the SVG files by email if that will help.

arnaudf-iter, 2015-08-05:
Hi, 
Please send the SVG files to frederic.arnaud@iter.org.
Thanks.

arnaudf-iter, 2015-08-05:
I can reproduce the error.
This is an internal error from Batik framework. There is a tag somewhere which does not belong with Batik and causes the error. It can be very tricky to find why...

A workaround that we use: 
Use the RSVG library tool to convert all tags to paths.
http://opensourcepack.blogspot.fr/2012/06/rsvg-convert-svg-image-conversion-tool.html

```
rsvg-convert.exe -a -z 0.9 -f svg <orginal_file> -o <output_file>
```

It will also reduce the SVG file size.
The only issue is that, by default, it will increase a bit the display size but you can use the -z option to get the right size (I put 0.9 as an approximation).

I tried with your file and it works for me. I'll send you the files by email.

gilesknap, 2015-09-24:
Sorry for the long delay. This did work for me. Thanks.

closed 2015-09-24 by willrogers