ControlSystemStudio/cs-studio #1303: OPI Display in Standalone Window issues

utzeln, 2015-08-18:
Making some tests, I found the following issues with OPI opened in a standalone window:

1/ Full Screen - Compact Mode affect CS-Studio editor part not the runtime window

2/ no zoom and OPI history toolbar - when the OPI is run using the green button, there is a new toolbar which is nice. This toolbar is not available in the standalone window

3/ cannot create snapshot for a new logbook entry

kasemir, 2015-08-18:
I believe this is by design. That standalone window mode is supposed to be just like EDM: No toolbar, no history, no .. DLS opens those windows via the --launcher.openFile mechanism and arranges them on the screen via X11 tools.

For SNS, we are not interested in using that mode. We have looked at disabling the 'standalone' mode via a preference setting, but there was no easy way to do this because it's added just like the other "Open with ..." options via plugin.xml entries.

What might be possible is move those plugin.xml entries into opibuilder fragments, so each site can then decide if their product should include them, or not.

willrogers, 2015-09-03:
1) I agree that the context menu options for compact mode and full screen should not be there for OPI shells.

2) This may be less relevant for people using OPI shells.  Certainly it is not how we expect the OPI shells to work.  This might be the reason that you want to disable the OPI shell functionality.

3) I am not familiar with this.  Is it a missing context menu option?

willrogers, 2015-09-03:
The other problem you mentioned is that there is no longer an option to open an OPI in a whole new Eclipse window.  See #544.