ControlSystemStudio/cs-studio #1304: Data browser widget in BOY

shroffk, 2015-08-18:
With the transition to 4.1.1 we want to ensure that most screens continue to work as closely as possible to the 3.3.x version.

1. The databrowser widget in BOY has its own show legend, show tool bar properties which are useful to ensure re-usability of the same .plt file in different opi screens. However it would be useful to have these properties are initialized for the first time to match the settings from the .plt file.

2. Since the trace name on Axis is a new feature it would be nice to have a property to enable disable this.

I can implement these changes unless someone has a better idea @kasemir @berryma4

kasemir, 2015-08-18:
Go ahead.

For what it's worth, the opiwidget doesn't work for me in the IDE.
Inside OpenDataBrowserAction, the call to ExceptionDetailsErrorDialog gets linked to the RAP version, resulting in a compile error.

It's OK for the tycho build from the command line.

shroffk, 2015-08-18:
Ok, There is no easy way to achieving the first point without a lot of hacking.
Should we comment out the legend visibility property for the time being?

This is another example where being able to link properties would prove very useful.

berryma4, 2015-08-19:
so, ... happened in the end?

berryma4, 2015-08-19:
So, people need to make a different databrowser file if they want a legend or toolbar?

shroffk, 2015-08-19:
- The opi widget just uses the show/hide property from the .plt file
- Users can still right click and use the show/hide legend/toolbar

- I attempted to have these properties separate but initialized from the .plt file, I attempted this by registering a listener to the file property which would then set the legend and toolbar properties...however I wasn't able to successfully resolve the relative file name. 
i.e. the resourcesHelper which correctly resolves relative paths in the run mode does not work in the edit mode (or am I missing something)

berryma4, 2015-08-20:
Can you setup a pull request to master too.

Thank you!

jbobnar, 2015-08-27:
@kasemir 
You should remove the plugin org.csstudio.rap.ui.util from your workspace. I keep two separate workspaces: one for RAP and one for RCP. This way you can control what is loaded during compile and runtime. The RCP workspace has no rap projects imported and RAP only has the plugins that the RAP applications need. Also make sure that the RCP doesn't have the RAP target defined.

closed 2015-09-29 by berryma4