ControlSystemStudio/cs-studio #1317: BOY - no more unique ID for each display $(DID)

utzeln, 2015-08-27:
To reproduce the problem BOY Examples/6_Macro.opi can be used as it displays always DID_0 even if the OPI is opened in different tabs.

kasemir, 2015-08-27:
I've never used anything that depends on the $DID.
How do you think it should work?

Every display gets its own ID, so it's for example a global `AtomicLong` that keeps incrementing? Displaying the same "display1.opi" in multiple UI parts -> Each has its down DID?

Or should it be based on the display path and UI part ID, so if you open "display1.opi" in some location, it gets a DID based on the hash of the path and the part ID. As you then navigate to sub-displays, the content of that part changes and the DID also changes, but when you navigate "back" to revert to "display1.opi", it has the same DID again?

Or should it be based just on the UI part ID, so if you open "display1.opi" , then navigate to a sub display, they all keep the same DID because it's just based on the UI part?

Or should the DID be based just on the display path, so "display1.opi" always has the same DID, no matter where it's opened?

willrogers, 2015-08-28:
I think this:

> Every display gets its own ID, so it's for example a global AtomicLong that keeps incrementing? 
> Displaying the same "display1.opi" in multiple UI parts -> Each has its down DID?

In fact 7cc4281b7ad32f9433ea978bdb1ec2eb46b6b50d had this effect, but it seems that it has been lost in a merge.

utzeln, 2015-08-28:
Agreed - every display gets its own ID - and yes, it was lost in a merge.

kasemir, 2015-08-28:
Will, thanks for fixing this twice!!

closed 2015-08-28 by berryma4