ControlSystemStudio/cs-studio #1320: BOY Text input widget submits the text value only after pressing Enter

utzeln, 2015-08-28:
Extract of the online documentation of the boy widget text update:

  At runtime, clicking the widget to edit text. Press ESC key to abort editing. 

  If there is no PV Name set for the widget, lost focus or clicking other place
  on the OPI will set the text. If PV Name is set, it requires explicitly hitting
  Enter or Ctrl+Enter to write the PV. 

  If Multi-line Input property is no, press Enter key to submit text. If
  Multi-line Input property is yes, press Ctrl+Enter to submit text. 

  Note: Even if you did not change the text, press Enter or Ctrl+Enter will still
  write the text to PV.

In ITER context, we do not write directly the PV but some local variable and then once the operator clicks on a Submit button, the entries are validated and submitted. This allows us to display clear and descriptive message when the validation is not OK and also to enter multiple values and submit them at once.

So, it is not obvious for the operator to press Enter on each entry fields and then click on Submit - he usually enters values, clicks Submit and sees the old values appear on the OPI without realising that he has just sent them again to the process.

One option would be to enlighten the fact that the text input is not final by changing the background colour for instance.

Another option proposed by Jaka is to define a boolean property "confirm at focus lost". By default it
is set to false; but if the value is true, the widget would set the value on the PV whenever it loses focus.

kasemir, 2015-08-28:
Depending on who you ask and when, you'll get different ideas.

a) Just enter a value. PV should update 'as you type'. Could be as extreme as a value going "1", then "13" as you type the number 13.
b) PV should update as you leave the widget.
c) PV should only update when you confirm with 'Enter'.
d) Type as long as you want, then there's a "Submit" button that submits everything on the page.
e) .. after you acknowledge an "Are you sure?" dialog.

Requiring 'Enter' on each value has been a good middle-way.
It's a control system, where entering values causes something to happen.
Requiring 'Enter' is a small and practical way to assert that you're thinking about the consequences a little bit. MEDM, EDM, now BOY have all used that mode successfully for decades.

At SNS we also have a few examples where several values all need to be set or adjusted, and then the overall update should take effect once all the parameters are in place.
That seems to match what you have in mind. Cannot do anything 'right away' on a single value, because other parameters may also need adjustment before starting the overall process.
That's handled via several individual PVs plus a "Submit" type PV which then performs the overall action on the IOC.
We find that consistency is still important: Each individual text field still requires 'Enter'. Since there is one more 'Submit' step, you could skip requiring the 'Enter', but having consistency is much more important to us. Having operators, especially short-term visitors on a beam line remember where 'Enter' is required and where not would be too confusing. Changing the background colors and then having to remember which play of colors indicates which type of entry mode doesn't really help.
Just 'Enter in every text field to confirm' is much more straight forward.

utzeln, 2015-09-04:
Thanks for the feedback.

We will implement this requirement in an ITER specific text field input control widget:

1. new property "background focus colour" to change the background color when the user clicks in the input field

2. new property "confirm at focus lost" [yes/no]  - if yes, the widget would set the value on the PV whenever it loses focus. If no, only after pressing Enter, the value is set to the PV.

closed 2015-09-04 by utzeln