ControlSystemStudio/cs-studio #1323: How to display a 1-d array PV ?

firstzp, 2015-09-01:
Hello, how to display a 1-d array PV in CSS?
I have a 1-d array PV and it can be changed timely. If I use X-Y Graphy widget  to display it, it can't display correct when the value is changed.

firstzp, 2015-09-01:
[image]

berryma4, 2015-09-01:
What version of CS-Studio are you using?
Is BOY using pvmanager as the pv connection layer?
(org.csstudio.opibuilder/pv_connection_layer=pvmanager)

berryma4, 2015-09-01:
Actually, I just tried this.  In probe, please show "metadata" from the drop down in the right corner.
I bet it changed from a VDoubleArray to a VString

berryma4, 2015-09-01:
I may be wrong, @shroffk, but I don't think probe supports writing to an array.
But, if you use the "Array" widget in BOY, you should see the graph responding to changes of the local variable.

shroffk, 2015-09-01:
What is not displayed correctly?
Are the values not changing?

Try this example and let me know if you still have problems
https://dl.dropboxusercontent.com/u/41120817/array.opi

berryma4, 2015-09-01:
I believe the problem is, people assume the format for entering arrays in Probe is how it is displayed (ie. [1,2,3,4,5]), but this is interpreted as a VString.

shroffk, 2015-09-01:
It seems like the array is entered correctly
loc://test(1,2,3)

berryma4, 2015-09-01:
In probe, look at the details.  After initialization it says vdoublearray.
If you enter [1,2,3] in the probe new value text box, the type changes to a
vstring.

shroffk, 2015-09-01:
Ok,
In any case, the way I use 1-D arrays is as shown in the example....
you can change the values
add/remove values
The plot keeps updating

firstzp, 2015-09-02:
It can correct display the value right now!
Thank you for your help!
[image]
Also, the buffer size of X-Y Graph and the size of 1-D array PV should be the same. And it can correctly update when the value of 1-D array PV changed.

firstzp, 2015-09-02:
The problem may be the format of probe!

shroffk, 2015-09-02:
Great
1. Displaying and writing to 1-D array works in BOY (XYGRaph, line plot, array....)
2. Probe does not support writing to Arrays - creating a new ticket ()

closed 2015-09-02 by shroffk