ControlSystemStudio/cs-studio #1329: Can't drag tabs between two open CS-Studio windows

willrogers, 2015-09-03:
I expected that if I had two CS-Studio windows open I would be able to drag tabs between them.  The tabs don't seem to recognise the second window.

Is this the expected behaviour?  Or is something strange happening in my build?

willrogers, 2015-09-03:
I see that this happens in Eclipse too, so there may not be much we can do about it.

kasemir, 2015-09-03:
Right now neither Eclipse nor CSS allow this:

1) Create new window (via Menu Window/New New Window, or in the case of CSS via right-click on action button, "Open in New Workbench")
2) Drag tabs between those windows.

You can drag 'editors' between those windows, but not 'views', and since *.opi runtimes are now views, ..

What you can do:
1) Drag a 'view' (opi runtime, ...) out of a workbench window so that it's detached
2) Drag tabs between the original workbench window and that detached part stack

So overall it's workable. Somebody who's a little familiar with these quirks can arrange windows as desired, then "Save As" on the perspective. But a new user who simply opens windows and drags stuff around will certainly run into several unexpected problems like this one.

willrogers, 2015-09-03:
OK - while this may not be desirable there's not much we can do @gilesknap.

closed 2015-09-03 by willrogers