ControlSystemStudio/cs-studio #1330: Multistate Symbol Monitor widget iniatilisation

charlesHenryPATARD, 2015-09-07:
Hi all,
I've some trouble with Multistate symbol Monitor widget. I think it's during the initialisation of widget. Sometimes the initial picture representing the state is not the good one. If I refresh the OPI the state change without any PV state changes. On the other hand, if the PV state change the picture and the state associated are always good.
We're using CSS in version 3.2.10.

Just below the terminal stack : after many refresh of OPI file
  1rst Refresh -	CommonMultiSymbolFigure.paintFigure() Picture profilHorizontal_0.svg // State HORS
                        CommonMultiSymbolFigure.paintFigure() Picture profilHorizontal_1.svg // State EN

  2nd Refresh - CommonMultiSymbolFigure.paintFigure() Image profilHorizontal_0.svg // State HORS
                         CommonMultiSymbolFigure.paintFigure() Image profilHorizontal_0.svg // State EN

Graphics behaviours,
1rst Refresh - 
[image]
2nd Refresh - 
[image]

Thanks a lot

charlesHenryPATARD, 2015-09-14:
Hi,
Someone can help me ? Who is in charge of Boolean and Multistate widgets ?

utzeln, 2015-09-14:
Sorry for the late answer. 
Could you please send me your test OPI? 
Are you using simulated PVs or EPICS PVs, as in previous versions we had an issue with simulated PVs as the enumerated values were loaded one by one and assigned somehow randomly, depending on which value was retrieved first.

charlesHenryPATARD, 2015-09-14:
Hi,
The opi file in attachment.

charlesHenryPATARD, 2015-09-14:
Hi,
Thanks for your answer. I'm using enum EPICS PV.
caget -d31 LBE2-PR12:InsertionState
<pre>
LBE2-PR12:InsertionState
    Native data type: DBF_ENUM
    Request type:     DBR_CTRL_ENUM
    Element count:    1
    Value:            En
    Status:           NO_ALARM
    Severity:         NO_ALARM
    Enums:            ( 4)
                      [ 0] Ni En - Ni Hors
                      [ 1] En
                      [ 2] Hors
                      [ 3] En ET Hors !!!
</pre>

I've send you the test opi by mail.

berryma4, 2015-09-29:
@charlesHenryPATARD ok to close?

charlesHenryPATARD, 2015-09-29:
Hi,
We just solved the bug today. It's was due to a bad map filling during the initialisation of the widget. @utzeln sould be push on GitHub the code correction as soon as possible.