ControlSystemStudio/cs-studio #1332: Data browser Axis labels

shroffk, 2015-09-09:
With the new Databrowser the aim is to maximize the plot area.

There are a few properties which would significantly improve that:

1. Allowing the Time Axis Title to be hidden would save a lot of wasted space as show in the example below, the axis labels make the "Time" title quite redundant, having a checkbox in the Time axis tab to show or hide this title could save a lot of screen space.

Merging the Font of all the Axis though convenient makes it difficult to customize the plot.
[image]

2. Allowing each Axis to have its own font would allow screen developers to select the most appropriate font based on the Axis Title, A Axis with many traces and a long name could be configured to have a smaller font

shroffk, 2015-09-09:
[image]

The Time Axis Title can be toggled on or off
[image]

Each Axis has its own label and scale fonts
[image]

kasemir, 2015-09-09:
How about just removing the Time axis title? No option, it's simply gone?

shroffk, 2015-09-09:
I think Removing Time Axis Title complete is great....it would also make the code much cleaner.

willrogers, 2015-09-10:
I just asked a couple of users and we couldn't see any need for a label saying 'Time' on the axis.

berryma4, 2015-09-10:
Sounds great to me.

berryma4, 2015-09-10:
Can we have a pull request for master too

shroffk, 2015-09-10:
Both this and #1304 should be merged but have conflicts.
Is there any plan to merge the bug fixed and additions from 4.1.x to master branch?

kasemir, 2015-09-10:
Yes, there are merge conflicts.

My leaning was cherry-picking whenever I did something to one branch to get that onto the other, then fixing the conflicts for just that, instead of trying to merge all of 4.1.x back to master.

shroffk, 2015-09-21:
Fixed and merged in to both 4.1.x and master

closed 2015-09-21 by shroffk