ControlSystemStudio/cs-studio #1333: OPI Display (Workbench) issues

utzeln, 2015-09-10:
Here are the results of the tests done using the BOY Examples:

1.	Boy Examples installed - test of workbench
[image]
 
2.	The opi run at the BOTTOM near the Console view when CS-Studio perspective is opened, or on the RIGHT in another tab after Properties when OPI Editor or Scan Editor perspectives are opened…
[image]
 
3.	On Widgets, clicking on Go Home button on the top left of the screen, will move the OPI on the LEFT near the Navigator view
[image]
 
4.	On Actions, open an OPI in a new tab, will run the opi on the LEFT near the Navigator view
5.	On Actions, open an OPI on the left, right, top or bottom, will always run the OPI on the LEFT near the Navigator view – Idem for Open OPI as Default view or Detached view. Also same thing executing an action from script: open OPI, it will run on the LEFT
6.	Making a Reset of CS-Studio perspective will rearrange but also close all running OPIs
7.	On Script – use Jython script, using the knob widget will open the console view on the LEFT
8.	From Widgets, clicking on a button will open the OPI on the LEFT

What is the rationale behind the opening of a view once the main opi is opened with "OPI Display (Workbench)"?

kasemir, 2015-09-10:
You need to run all your tests in the OpiRuntime perspective.
It's the only one that has the magic placeholders to support LEFT/RIGHT/.. positioning.

shroffk, 2015-10-07:
Adding a simple fix to the default "CS-Studio" perspective so that the "open in Workbench" opens in the center of the screen.

willrogers, 2015-10-29:
The only way I can find to sensibly run the BOY Examples is as follows:

* Reset OPI Runtime perspective
* Window -> Show View -> Navigator
* Right click `main.opi` and Open in Workbench
* *minimise editor area that just opened!*

Now you're in the right perspective and the OPIs will open in the correct places.  However, it would be nice if this all just happened for you.