ControlSystemStudio/cs-studio #1343: o.c.alarm.beast.ui: add alarm_disable to AuthIDs

berryma4, 2015-09-14:
I would like to add another auth for disabling/enabling alarms.
alarm_disable
```java
public interface AuthIDs
{
    /** ID used to (un-)acknowledge alarms */
    public static final String ACKNOWLEDGE = "alarm_ack";

    /** ID used for changes to the alarm configuration */
    public static final String CONFIGURE = "alarm_config";

    /** ID used to disable and enable alarms */
    public static final String DISABLE = "alarm_disable";
}
```
I'm assuming I will need to add the check to 
DisableComponentAction

and modify the GUI class in o.c.alarm.beast.ui.alarmtable to show the option when authed.

Anything else?  Or disagree?
@kasemir

kasemir, 2015-09-14:
In principle, that sounds good. It is already auth-sensitive, using CONFIGURE, so I guess you only need to check DisableComponentAction & EnableComponentAction and where they're used.
Then also mention the new alarm_disable in o.c.security/authorization.conf 

After the update, this will require everybody to add the new alarm_disable to their actual production authorization.conf 
Fine, not a big deal.

Quirk:
You can also enable/disable an alarm via the configuration dialog.
That was actually the original way to do that.
The enable/disable shortcut in the context menu simply does that for a complete subtree.

So you could end up in a situation where enable/disable is prohibited.
.. for example simply because you didn't update your production authorization.conf to add people to that new authorization.
.. but you can still enable/disable via the configuration dialog, if you have that authorization.

So ideally it would work like this:
If you have alarm_config, you also gain alarm_disable authorization.
Without alarm_config, you could still have alarm_disable authorization.
But that would add a lot more logic to the o.c.security plugin.

berryma4, 2015-09-29:
I like your "ideally" scenario.

closed 2018-06-12 by berryma4