ControlSystemStudio/cs-studio #1347: caj array write limit issue

berryma4, 2015-09-21:
Someone here wrote a BOY screen that would write large arrays to a pv (for a function generator). 
But, we ran into this limit:
```java
java.lang.RuntimeException: Write failed
	at org.epics.pvmanager.PVWriterDirector.syncWrite(PVWriterDirector.java:332)
	at org.epics.pvmanager.PVWriterImpl.write(PVWriterImpl.java:148)
	at org.csstudio.simplepv.pvmanager.PVManagerPV.setValue(PVManagerPV.java:422)
	at org.csstudio.opibuilder.scriptUtil.PVUtil$2.run(PVUtil.java:346)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:54)
Caused by: java.lang.RuntimeException: Message exceeds write buffer size (com.cosylab.epics.caj.impl.CachedByteBufferAllocator.buffer_size)
	at com.cosylab.epics.caj.impl.CATransport.submit(CATransport.java:892)
	at com.cosylab.epics.caj.impl.requests.AbstractCARequest.submit(AbstractCARequest.java:88)
	at com.cosylab.epics.caj.CAJChannel.put(CAJChannel.java:994)
	at com.cosylab.epics.caj.CAJChannel.put(CAJChannel.java:840)
	at org.epics.pvmanager.jca.JCAChannelHandler.put(JCAChannelHandler.java:299)
	at org.epics.pvmanager.jca.JCAChannelHandler.write(JCAChannelHandler.java:582)
	at org.epics.pvmanager.DataSource$5.scheduleNext(DataSource.java:374)
	at org.epics.pvmanager.DataSource$5.run(DataSource.java:405)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)
Caused by: java.nio.BufferOverflowException
	at java.nio.DirectByteBuffer.put(Unknown Source)
	at java.nio.DirectByteBuffer.put(Unknown Source)
	at com.cosylab.epics.caj.impl.CATransport.submit(CATransport.java:890)
	... 10 more
```

We seem to be running into a property that is not configurable in the CS-Studio gui.   http://grepcode.com/file/repo1.maven.org/maven2/org.epics/caj/1.1.10/com/cosylab/epics/caj/impl/CachedByteBufferAllocator.java#CachedByteBufferAllocator.0bufferSize

If you add this to your C:\Program Files\cs-studio\cs-studio.ini file all is well:
```ini
-Dcom.cosylab.epics.caj.impl.CachedByteBufferAllocator.buffer_size=40000000
```

berryma4, 2015-09-29:
Because preferences change in diirt 3.  @shroffk do we just add this to diirt config files?

shroffk, 2015-09-29:
At least for the initial 4.2 release this will be configurable through the diirt/config

berryma4, 2015-11-17:
closing ticket because there is a way to set the property using java properties, although not ideal.
I made a diirt ticket, such that this may be configured with the diirt xml file.
https://github.com/diirt/diirt/issues/63

closed 2015-11-17 by berryma4