ControlSystemStudio/cs-studio #1356: Release 4.2.x (master)

berryma4, 2015-09-28:
Proposing to freeze the master branch to 4.2.x.

  - [ ] BOY Perspectives #544
  - [ ] Diirt 3 #1171
  - [ ] Eclipse 4.5 #1064
  - [ ] report BOY regression tests (ControlSystemStudio/opi-regression-tests)
     https://docs.google.com/forms/d/1cKILK51oWpVcy5iDCe4gPK6mvCSvlMWIS5JazuO2_Lc/viewform
  - [ ] make 4.2.x branch
  - [ ] make new github milestone 4.3.0 - testing (master)
  - [ ] make new github milestone 4.2.1 - stable (4.2.x branch)
  - [ ] make Jenkins jobs (turn on testing!)
  - [ ] make Jenkins csstudio.product release
  - [ ] run Jenkins csstudio.product release

Site Representatives, are there any other outstanding issues that need to be addressed before release of 4.2.x?
@berryma4, @kasemir, @shroffk, @richardfearn, @willrogers, @utzeln, @mjclarke01, @jpenning, @ksdj

willrogers, 2015-09-29:
Three comments from me:

* #544 I'm sure we still don't have the ability to save and share perspectives.  That's OK, it's just a reasonably-sized piece of work that no-one has managed to do.
* #1171 The one key configuration option we need available is to change EPICS ports.  Is that going to become tricky?
* I have one more possible feature relating to OPI shell that I could push before the freeze.

berryma4, 2015-09-29:
for #544, maybe we should make a new ticket for the save and share part.

willrogers, 2015-09-29:
Agreed.

kasemir, 2015-09-30:
The code for #544 is on the master branch, it compiles, it runs, so in that sense it's OK to create a 4.2 release off master.

I'm afraid, however, that it's not a perfect release for end-users.

As explained in the how-to comment https://github.com/ControlSystemStudio/cs-studio/issues/544#issuecomment-107684346 , you're now best starting with the OpiRuntime perspective. When you do, newly opened views will appear as usual in the 'center', unless they're specifically configured to appear left/right/top/bottom. If you don't start there, you get somewhat unexpected results compared to previous behavior (#1333, emails I got from INFN).

What might help: Base the default CS-Studio perspective that most sites have as their initial perspective on the OpiRunner, i.e. add the opiViewLEFT/RIGHT/TOP/BOTTOM perspective folders to it.
Or add the Navigator to the OpiRuntime perspective, which means it can pretty much _be_ the default perspective for many sites.

Should we have a hangout on this, upcoming Monday Oct. 5? Or dedicate time at the regular monthly CSS hangout, next Oct. 7?

willrogers, 2015-09-30:
I would come, but I'm away next week.

berryma4, 2015-09-30:
I agree, this is not that greatest release for end users.  It seems we are starting to follow the Windows OS pattern (http://www.urbandictionary.com/define.php?term=Windows+Pattern).

berryma4, 2015-09-30:
I rather have a Monday Oct 5 meeting for this.
The Oct 7 regular meeting, I would like to concentrate on items for the next release, and try to adjust our focus to the customer a little more.  We have been spending a lot of time on the build, diirt upgrade, cleaning up code, unit test, etc..  These of course make the end product better, but we should think about changes we can make that will make the end user ecstatic.  Remember, we can't judge our our work.

Two examples come to mind to avoid:
  - Myspace fit a need everyone wanted, but the user experience was awful.
  - Google Wave was technically sound, but nobody knew how to use it or why they should.

kasemir, 2015-10-05:
BOY Perspective: In google hangout this morning we decided that the "Show OPI Runtime Stacks" preference should appear in the UI. Turns out it does already:

[image]

That preference can be changed at runtime, and a "Reset Perspective" on the OPI Runtime perspective will then show not not show the placeholders that can help with arranging the layout.

[image]

berryma4, 2015-12-01:
While attempting the BOY regression tests, I had the following issues:
@shroffk @willrogers @kasemir @utzeln 
- [ ] #1483 BOY Examples/widgets/4_1_GroupingContainer.opi
- [ ] #1484 BOY Examples main.opi
- [ ] #1485 "Name of user" lower rectangle too small to read username
- [ ] #1486 opi-regression-tests - RulesTests - General_rule.opi
- [ ] #1487 opi-regression-tests - RulesTests - Switch_rule.opi
- [ ] #1488 opi-regression-tests - WidgetTests - ActionButton_test.opi

berryma4, 2015-12-22:
https://docs.google.com/forms/d/1cKILK51oWpVcy5iDCe4gPK6mvCSvlMWIS5JazuO2_Lc/viewanalytics

| Test       | Result           |
| ------------- |:-------------:|
| timestamp | 12/22/2015 12:43:42 |
| issue ticket | https://github.com/ControlSystemStudio/cs-studio/issues/1540 |
| Tester | berryman |
| __BOY Examples__ |
| All BOY Examples | Pass |
| __Issues__ |
| Issue 1081 | Pass |
| Issue 1103 | Pass |
| Issue 1197 | Pass |
| __Performance Tests__ |
| graph_performance_test.opi  | Pass |
| large_led_performace_test.opi | Pass |
| small_led_performace_test.opi | Pass |
| tank_performace_test.opi | Pass |
| test_update_performance_test.opi.opi | Pass |
| __Rules Tests__ |
| General_rule.opi | Pass |
| Graphs_rule.opi | Pass |
| LED_color_rule.opi | Pass |
| Label_rule.opi | Pass |
| Switch_rule.opi | Pass |
| __WidgetTests__ |
| ActionButton_test.opi | Pass |
| BubbleGraph_test.opi | Pass |
| ComboBox_test.opi | Pass |
| Container_test.opi | Pass |
| Gauge_Meter_Thermometer_Tank_test.opi | Pass |
| LED_test.opi | Pass |
| TextInputWidget_test.opi | Pass |
| TextUpdate_test.opi | Pass |
| XYGraph_test.opi | Pass |

closed 2015-12-22 by berryma4

berryma4, 2015-12-22:
http://download.controlsystemstudio.org/release/4.2/