ControlSystemStudio/cs-studio #1384: PV preference page fails

berryma4, 2015-10-12:
[image]

berryma4, 2015-10-12:
@shroffk

kasemir, 2015-10-12:
That would make sense, because I see a build error in code mentioned in #1171 where the pvmanager.ui cannot be built.

shroffk, 2015-10-12:
This is strange, are you building all of core or just the utility module

kasemir, 2015-10-12:
>This is strange, are you building all of core or just the utility module

I use a composite repo

https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/css_repo
that lists the locally built repository/target/repository for maven-osgi-bundles, 3rd party, core, applications, and then build all like this:

(cd maven-osgi-bundles; mvn clean verify)
(cd cs-studio-thirdparty; mvn clean verify)
(cd cs-studio/core; mvn clean verify)
.. and that last step fails with

[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT @ /home/ky9/git/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/pom.xml
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier requires 'package org.diirt.vtype 3.0.0' but it could not be found

shroffk, 2015-10-12:
Does the remote diirt  p2 repository get added to the list of repos?
>> Adding repository http://diirt.download.controlsystemstudio.org/diirt/3.0.1

kasemir, 2015-10-12:
That's the issue for me!
core/pom.xml lists http://diirt.download.controlsystemstudio.org/diirt/3.0.1 as part of the cs-studio-sites profile, which I disable by defining the cs-studio property since I want to use only locally built CSS artefacts.

So I'd either have to build diirt locally, or list the remote dirt repo in my composite repo.

willrogers, 2015-10-12:
I find that the `core` composite repo hasn't been updated to include the `diirt` repo anyway...

kasemir, 2015-10-12:
I've updated the docbook: Full build needs to also clone diirt, build that, then list the diirt/p2/target/repository in the composite repo. With that, my cs-studio/core builds again.

So none of that helps at all with Eric's original question on the PV preference UI being broken.

shroffk, 2015-10-12:
We are being too proactive with fixing things :)
The core p2 repo description has a 2 entries for diirt... @berryma4 is fixing it

berryma4, 2015-10-12:
fixed.

shroffk, 2015-10-21:
Should this issue be closed now?

closed 2015-10-21 by berryma4