ControlSystemStudio/cs-studio #1387: Diirt update changed alarm JMS plugins

kasemir, 2015-10-12:
The diirt update https://github.com/shroffk/cs-studio/commit/95fce851bb9bc701e1d579957b0da14d08eae640 changed org.csstudio.alarm.beast.annunciator/META-INF/MANIFEST.MF from using org.csstudio.platform.libs.jms to javax.jms

The SNS product now fails to build.

@shroffk do you remember why the JMS bundle was changed? The diirt update should not require it.

kasemir, 2015-10-12:
org.csstudio.platform.utility.jms/META-INF/MANIFEST.MF also requires org.csstudio.platform.libs.jms, but it looks like we optimized that out.

If we use javax.jms instead of org.csstudio.platform.libs.jms, org.csstudio.platform.utility.jms will also need to be updated. Unclear why it still succeeds to build.

shroffk, 2015-10-12:
I think there is a version of the old plugin in the third part plugin repository.

Would if be possible to use javax.jms plugin from orbit instead of wrapping it in platform.libs.jms. 
We could also get the apache.commons packages from bundles already in orbit.

I can't find the activemq bundles, but @berryma4 could we add the activemq bundles to the https://github.com/ControlSystemStudio/maven-osgi-bundles/blob/master/pom.xml
which should download the maven plugins and wrap them into osgi bundles and publish to an update site. This way we do not have to copy binaries into the repository and we can easily mange versions by simply making minor changes in the pom files.

The only thing we would have to provide is the preference for these libraries. 

@kasemir I am sorry about the changes breaking your product. We can revert the dependencies back to include platform.libs.jms and since this plugin is still there in the third party repo the build should be back.

berryma4, 2015-10-12:
I'll add whatever needed to maven-osgi-bundles.

kasemir, 2015-10-12:
Yes, for now I'd like to have org.csstudio.alarm.beast.annunciator again depend on platform.libs.jms. I'll send a pull request, because that way I get my products to build, while it fails with the javax.jms dependency.

Stepping back, I guess I don't understand what you want to change. We removed the JMS/ActiveMQ binaries from our source repo. The platform.libs.jms plugin is now assembled in the 3rd party repo, fetching the JMS/ActiveMQ binaries form somewhere. That seems good: No binaries in the repo, yet we still get one libs.jms plugin that contains all we need. Do you want to split that into individual plugins for just the JMS interface, the activemq implementation, xstream for JMS-over-HTTP and so on?

closed 2015-10-12 by berryma4

shroffk, 2015-10-13:
1. The javax.jms will have to include in some feature which is included in the product.

2. When I say it should be split into individual bundles, we do not need to manually create separate plugins. By correctly setting up the maven-osgi pom all the bundles become available via that p2 sites (most of them are already available in orbit). Each application developer/product manager then decides which jms providers should be included in the application feature and/or their product.
To me this seems better than have plugins will all possible implementation.

PR #1388 (2015-10-12): alarm annunciator: Revert to platform.libs.jms
Was changed to javax.jms, which would require more updates to for example SNS products

Fixes #1387