ControlSystemStudio/cs-studio #1389: Combo widget with numbers won't properly set <VString>

NickeZ, 2015-10-13:
I cannot use a combo widget with items = [1,2,3,...] to set a pv loc://test<VString>. Is there any known workaround for this?

kasemir, 2015-10-13:
This may be related to the fact that local PVs do not support enumerated values.
See https://github.com/diirt/diirt/issues/62

NickeZ, 2015-10-13:
It might be, but to extend my bug report: This only happens when the items are *numbers*. So if the items were [Ch1, Ch2, Ch3...] it would properly set the <VString> PV.

willrogers, 2015-10-13:
I can reproduce this.  If nothing specifies the type of the local PV, it will become a VDouble and show up as 1.0, 2.0, 3.0.  If you specify it as VString the writes don't take effect.

Probably related to https://github.com/diirt/diirt/issues/59 as well.

willrogers, 2016-01-07:
I believe this pull request would resolve this problem:

https://github.com/ControlSystemStudio/diirt/pull/14

willrogers, 2016-01-27:
@NickeZ can we make this more explicit?  In my tests the Combo will set the local PV to 1.0, 2.0 etc, which you can read using a TextUpdate.  Does this work for you?

If the PV type is specified as <VString> it just won't work.

The fix will allow it to work as a VString.

NickeZ, 2016-03-01:
I used my workaround since this didn't work so I can't really test it again now.

The issue was that integers were printed as floats and there was no way of specifying strings, like ["1", "2", "3"], IIRC. I couldn't use floats because I used the final value to compose a PV name and the PVs where named "prefix":ch0:"something".