ControlSystemStudio/cs-studio #1390: Problems on new master branch

willrogers, 2015-10-13:
I've built the DLS product against the new master branch.  I have two strange problems:

* The cross on the top right of the window is not closing the window.
* Using the EPICS PV Tree view throws a `NoClassDefFoundError`:

```
2015-10-13 17:16:32.729 SEVERE [Thread 1] org.csstudio.logging.PluginLogListener (logging) - Unhandled event loop exception
java.lang.NoClassDefFoundError: org/diirt/datasource/expression/SourceRateExpression
	at org.csstudio.diag.epics.pvtree.PVTreeModel.setRootPV(PVTreeModel.java:67)
	at org.csstudio.diag.epics.pvtree.PVTreeView.setPVName(PVTreeView.java:232)
	at org.csstudio.diag.epics.pvtree.PVTreeView$1.handleEvent(PVTreeView.java:131)
	at org.eclipse.swt.widgets.EventTable.sendEvent(EventTable.java:84)
	at org.eclipse.swt.widgets.Display.sendEvent(Display.java:4481)
```

These may well be due to idiosyncrasies in our build in which case I apologise.

kasemir, 2015-10-13:
I can confirm that the SNS product also has problems, but different.
In our case, the PVManager isn't initialized because I yet need to add the new config file directories for diirt.
The original org.csstudio.diag.epics.pvtree still works fine because it's not using the PVManager. Your product must include the org.csstudio.diag.epics.pvmanager.pvtree plugin.

willrogers, 2015-10-13:
For the window closing, I find that the problem is at https://github.com/ControlSystemStudio/cs-studio/blob/master/core/diirt/diirt-plugins/org.csstudio.diirt.util.preferences/src/org/csstudio/diirt/util/preferences/DiirtStartup.java#L46.  

Because this method returns false, the window doesn't appear to close.  Is anyone else seeing this problem?

willrogers, 2015-10-13:
There's also a syntax error in the `.classpath` file at https://github.com/ControlSystemStudio/cs-studio/blob/master/core/diirt/diirt-plugins/org.csstudio.diirt.util.preferences/.classpath.

One of the XML attributes hasn't got its closing quote.

berryma4, 2015-10-13:
I get the same issue, where the upper right close x button current work.

berryma4, 2015-10-14:
I can close workbench windows again!!!  Thank you!

willrogers, 2015-10-15:
@kasemir we've put the diirt config inside the `org.csstudio.dls.product` plugin alongside BOY configs.

There's one hitch - Diirt doesn't understand the `platform:` URIs that Eclipse uses.  However, it's possible to adapt this inside @shroffk's preference plugin.  I've got a commit to do that which I'll push if everyone's happy: https://github.com/dls-controls/cs-studio/commit/75b865c994e73690d3e4c882d6b44f1893e2bd90

There's another thing, which is that I expect Diirt won't find the files if they are inside a jar.  I found that `org.csstudio.dls.product` already had the configuration `Eclipse-BundleShape: dir`, which keeps the plugin as a directory and allows everything to work.

Thoughts?

willrogers, 2015-10-15:
The EPICS preference page is still present.  It is contributed by the plugin `org.csstudio.platform.libs.epics.ui` which is included in the feature `org.csstudio.core.platform.feature`.

shroffk, 2015-10-17:
@willrogers you can go ahead and make the pull request to support the platform urls in the diirt preferences.

Regarding where to put the diirt folder, I use the product.configuration feature for the diirt folder...using the root folder which ensures that the preferences folder gets installed in $ECLIPSE_HOME

shroffk, 2015-10-17:
@kasemir is it ok if we remove the platforms.libs.epics from the feature?

willrogers, 2015-10-20:
@shroffk I'm happy to use your method for storing the configuration inside the product, but I need more details.  Do you have an example on Github?

shroffk, 2015-10-21:
Ok,

I think right now it is hidden in some issue ticket but I should move it somewhere more accessible

Kunal

Sent via the Samsung GALAXY S®4, an AT&T 4G LTE smartphone

willrogers, 2015-10-22:
I found this configuration method on the `org.ccstudio.product` repo.  What can you put in your configuration file to specify $ECLIPSE_HOME?

I find that the configuration is passed directly to diirt with no interpretation.  This means only absolute paths or relative paths (but you have to launch CSS from the correct directory) are working.

shroffk, 2015-11-17:
Should this be closed?

willrogers, 2015-11-17:
I think so.

closed 2015-11-17 by berryma4