ControlSystemStudio/cs-studio #1398: Save and share perspectives

willrogers, 2015-10-20:
This was one of the original goals of #544, but all other aspects of that ticket are complete.  Using views for OPIs enables more control over perspectives.

Kay's original description:

> Goal is to be able to “save” the layout of *.opi displays, so it can be “restored” and operators can switch between different layouts.
> Maybe including a “frozen” mode where a “restored” layout cannot be altered.

Developers at Diamond are keen to be able to share perspectives between different workspaces.

willrogers, 2015-10-20:
Copied from #544, by @kasemir:

Some clarification what I think this is about:

> Users want to save and restore the layout of *.opi panels on the screen.
> As an example, the current layout may be called "Vacuum". "VacPumps.opi" and "VacValves.opi" are open in some arrangement on the screen.
Then operators can switch to a "Cooling" layout that instead opens "WaterPumps.opi" and "WaterTemperatures.opi" in some other arrangement on the screen.
Operators can switch between the two layouts.

> I still think that these layouts can just be the RCP Perspectives, as also suggested by #438. Changing the opibuilder.Runtime to exclusively use "Views" instead of "Editors", and fixing how those "Views" keep their *.opi file path (#674) should mostly get us there. Unclear if perspectives would be configured by adding some new XML format file reader as per #438, or we study how perspectives are currently persisted and somehow create those files, or only use "Save As" and then copy the persisted perspectives from a development computer to the control room.

> "Frozen" has been brought up in that operators should not be allowed to move things around or close windows. The RCP API for the Perspective offers IPageLayouts.setFixed(true), but that actually no longer functions in E4 (https://bugs.eclipse.org/bugs/show_bug.cgi?id=331402). Realistically, I consider this a weak requirement. EDM etc. never prevented window movement. The web or email browser in which you read this can be moved. Operators will certainly need to open a Data Browser, start a logbook entry, maybe move windows around to better see what they need to see.
> A more realistic requirement is a "Reset": In case operators moved windows or for example closed the "WaterPumps.opi", a "Reset" of the "Cooling" perspective should restore it to the original layout, re-opening the "WaterPumps.opi". Fundamentally, perspectives already support "Reset", so that should not be a technical problem.

willrogers, 2015-10-20:
This Eclipse bug is related: https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811

willrogers, 2015-10-20:
Copied from #544, by @kasemir:

### Snapshot, Restore complete Workbench

We don't have a nice way to save a selected perspective so that it could be added to another CS-Studio instance, for example to prepare perspectives for a beam line.
Will's comment ~~from April 28~~ above points to a bug report which shows activity, but it's not solved.

Meanwhile, we know that the workspace file
   
    .metadata/.plugins/org.eclipse.e4.workbench/workbench.xmi

contains the definition of all perspectives (with list of their view) and views (with their memento for *.opi path and macros).

While CS-Studio is _not_ running, you can copy that file, then either restore it later to reset a desired workbench setup,
or place it in the workspace of another CS-Studio instance, which is also not running at that time, to duplicate the
workbench setup.

berryma4, 2015-12-08:
Great news!
https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811
Verified in the Eclipse integration build I20151207-2000

berryma4, 2015-12-22:
Update:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811

Eclipse Genie CLA Friend 2015-12-22 12:26:40 EST
Gerrit change https://git.eclipse.org/r/63125 was merged to [R4_5_maintenance].
Commit: http://git.eclipse.org/c/platform/eclipse.platform.ui.git/commit/?id=4b17577b9a1a592c1f71a6a1756bb8d3c29b2cb1

berryma4, 2016-01-26:
Update:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811

(In reply to Eclipse Genie from comment #86)
> Gerrit change https://git.eclipse.org/r/58236 was merged to
> [R4_4_maintenance].
> Commit:
> http://git.eclipse.org/c/platform/eclipse.platform.ui.git/commit/?id=807e99692a618dde238605b07c75084a29137d60
> 

==> Marking as FIXED.

berryma4, 2016-03-01:
Update:

https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811

- Comment #94 from Thomas Schindl <tom.schindl@bestsolution.at> ---
Arghhhhhh - this breaks the widget independency of the e4 - you are adding
JFace/SWT to the central component of e4 and hence you break e(fx)clipse
runtime layer completely in a 4.5.x release!

http://git.eclipse.org/c/platform/eclipse.platform.ui.git/diff/bundles/org.eclipse.e4.ui.workbench/META-INF/MANIFEST.MF?id=807e99692a618dde238605b07c75084a29137d60

I'm speechless! if you are interested what this means for down stream you can
follow https://bugs.eclipse.org/bugs/show_bug.cgi?id=488730 but in boils down
to us not being able to adopt Mars.2 - This patch needs to be rolled back
please!

willrogers, 2016-03-02:
It looks like they're going to make a workaround so as not to disrupt people like us.

willrogers, 2016-04-06:
All done.

closed 2016-04-06 by willrogers