ControlSystemStudio/cs-studio #1406: Diirt Configuration

shroffk, 2015-10-28:
There are some issues regarding how to configure the new diirt datasources
- [ ] Diirt configuration preference page
- [ ] Help documentation is needed on how to configure diirt
- [ ] remove the references to the old pvmanager plugins and features and replace them with diirt plugins and features (https://github.com/ControlSystemStudio/org.csstudio.product/issues/14)

shroffk, 2015-10-28:
Created the Diirt preferences page which lets you define the location of the diirt configuration folder.

In addition it has a directory view which shows all the individual configuration files. Double click on any file will result in opening that file with the configured default editor (so you don't have to go searching for it).

[image]

[image]

shroffk, 2015-10-28:
Note:
@willrogers has already worked on supporting the platform URLS http://help.eclipse.org/juno/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Fguide%2Frcp_perspective.htm

Do we want to support these or use the format requested by @kasemir 
````
org.csstudio.diirt.util.preferences/diirt.home=/etc/diirt
org.csstudio.diirt.util.preferences/diirt.home=${user.home}/.diirt
org.csstudio.diirt.util.preferences/diirt.home=${osgi.install.area}/diirt
org.csstudio.diirt.util.preferences/diirt.home=${osgi.configuration.area}/diirt
org.csstudio.diirt.util.preferences/diirt.home=${osgi.instance.area}/diirt
````

kasemir, 2015-10-28:
Not sure about the rcp_perspective link you mention, but I'm fine with platform URLs.

They would cover what I really need: install location.
They'd also cover instance or config location.

They wouldn't allow specifying the user's home directory, but when we look for "specific to user" I think instance location (= workspace) is basically the same idea.

So: Yes, platform URLs are OK.

kasemir, 2015-10-28:
I found this link on platform URLs: http://help.eclipse.org/mars/index.jsp?topic=%2Forg.eclipse.platform.doc.isv%2Freference%2Fmisc%2Fplatform-scheme-uri.html

So `platform:/config/diirt` would allow placing the diirt config in `css_product/configuration/diirt`. There's no way to select instance (workspace).

shroffk, 2015-10-28:
No, it is not possible to get the workspace from the platform urls, but if you have a default workspace you could use something like ~/css-workspace

kasemir, 2015-10-28:
Stepping back, what do we need?

The plugin_customization.ini in the product plus the -pluginCustomization command line option worked fine, but now we have diirt I think I need these two alternatives:

* Install Location
I need to place the diirt root in the install location, because one download of CSS for SNS office users should include everything.

* Some absolute location in the file system
I need a way to modify this for installations on beam lines, which each have specific settings.
An absolute path like "/etc/diirt" would work.

The platform URL would allow `platform:/config/diirt`. The config dir is typically the `configuration` dir in the install location, but for a multi-user install where the install/configuration directory is read-only, Eclipse also uses $HOME/.eclipse for some portion of the configuration. So I'm not sure if `platform:/config/diirt` would always work for me.

* Config Location
Is that really needed?

* Instance or ${user.home}
Is that really needed?
If a user wants private settings, could always set the absolute location to somewhere in their home directory.

willrogers, 2015-10-28:
We are looking for a user-friendly solution, and of course none are as good yet as having port configuration in the preference page itself.  

We plan to have two configurations in the product: `diirt_5064` and `diirt_6064`.  This should satisfy the majority of our users.  If you find a preference that says `diirt_5064`, it's not too much to learn that a `diirt_6064` is there too, although trying different ports is going to fail.

Any more advanced users will have to copy the directories from their to a location of their choice and edit the configuration themselves.

Another option would be to copy the default configuration into someone's home directory.  I don't think this is necessarily a good idea, because then if they mess it up there's no simple way to revert.

willrogers, 2015-10-30:
Thanks for adding in the platform URI parts.

kasemir, 2015-11-02:
Since we agree on supporting the platform:.. URL, how will it be implemented?
In ..csstudio.diirt.util.preferences/src/org/csstudio/diirt/util/preferences/DiirtStartup.java as Will had demonstrated?

willrogers, 2015-11-02:
Kunal's already pulled in that change, so it should work.

kasemir, 2015-11-02:
As far as I can tell, only the `DiirtPreferencePage` was updated to understand "platform:..".
In the preference GUI, I can enter "platform:.." and it's transformed into the absolute path.

But I don't want every user to once manually enter "platform:.." in the preference GUI.
I want to set 
```
org.csstudio.diirt.util.preferences/diirt.home=platform:/config/diirt
```
in the product's plugin_customization.ini, and on startup the DiirtStartup code then needs to translate that it into the install location.

shroffk, 2015-11-02:
Done, This should be supported now

````
org.csstudio.diirt.util.preferences/diirt.home=platform:/config/diirt
````

kasemir, 2015-11-02:
Yes, that looks like what I need.

Regarding the duplication, when I manually type "platform:/config/diirt" in the preference GUI, I think I want "platform:/config/diirt" to be stored in the preferences as is, which is then resolved on the next startup by the code you just updated.
I don't want the expanded/substituted version to be saved in the preferences.

shroffk, 2015-11-02:
Both the startup support and the preference page store the url as is (with platform:\) and not in the expanded form.
It only gets substituted to an expanded path when setting the system.property or the File display dialog.

kasemir, 2015-11-02:
Thanks for clarifying, that sounds perfect. Unless Will finds a problem, we should merge this.

willrogers, 2015-11-03:
Works for me now.

berryma4, 2015-12-08:
Can I close this @shroffk

closed 2015-12-08 by berryma4