ControlSystemStudio/cs-studio #1408: OPIShellSummary breaks OPIRunner perspective

willrogers, 2015-10-28:
This is odd.  One of our users reported that when the `OPIShellSummary` view is opened, opening views in different stacks in the OPIRunner perspective stopped working - all new OPIViews are opened in the same stack.

I can fix this by removing the `OPIShellSummary` view from the category `org.csstudio.opibuilder.views`, but I don't understand *why* this works.

@kasemir, any ideas?

kasemir, 2015-10-28:
No, I don't understand.

The category `org.csstudio.opibuilder.views`, which shows up as "BOY" in the view listing, is used by all the "OPI View" views `org.csstudio.opibuilder.opiView`, `org.csstudio.opibuilder.opiViewLEFT`, ...

Your fragment _adds_ another view to that category. I would worry if the OPIShellSummary used a view ID that's already used by the OPI Views, but your view ID is unique.
One thing that might make a difference: All the other views in the `org.csstudio.opibuilder.views` category allow multiple instances, the OPIShellSummary doesn't. . which makes sense, but maybe there's a a quirk in Eclipse that doesn't allow this?

willrogers, 2015-10-29:
@shroffk makes a similar comment here; https://github.com/ControlSystemStudio/cs-studio/pull/1374

I can't tell if this is referring to OPIShells (the same problem) or workbench windows (a similar problem).

willrogers, 2015-11-05:
@shroffk can you confim whether this is fixed or not?  I believe it is associated with the new view as described above.

berryma4, 2015-12-08:
fixed?

willrogers, 2015-12-08:
Sort of.  I expect we will come back to this but you can close this ticket now.

closed 2015-12-08 by berryma4

shroffk, 2015-12-08:
Yes, the scope of this ticket has been address and resolved.
As will pointed out both this as diirt configuration might need more work later