ControlSystemStudio/cs-studio #1410: Opening OPIs from navigator opens editor pane

willrogers, 2015-10-29:
It's possible that everyone knows all this already, but:

If you right-click on an OPI in the navigator view, you have four options:

* OPI Display (Standalone Window)
* OPI Display (Workbench)
* OPI Editor
* Text Editor

Whichever you choose, if the editor pane is hidden it reappears.  If you choose OPI Editor or Text Editor this is correct, since they are editors.  If you choose either OPI Display, this is incorrect, since they are not.

This happens because we use the extension point `org.eclipse.ui.editors` in order to get the sensible behaviour from the navigator.  It may be very difficult to do anything about.

willrogers, 2015-10-29:
Possibly for this reason, neither of the main workflows we're planning to use at Diamond involves the user interacting with the navigator much.

kasemir, 2015-11-05:
I'm afraid we can't do much about this.
As you mention, we declare `org.csstudio.opibuilder.runmode.DisplayLauncher` as an __external__ "editor launcher" in `org.csstudio.opibuilder.rcp/fragment.xml` to handle `*.opi` files. In principle, those launcher are meant to start external tools. In their API, they only receive the path of the input file, no parent window or part stack.

But when I set a breakpoint in our `DisplayLauncher`, I find that it's called from within `org.eclipse.ui.internal.WorkbenchPage` like this:

````
// line 3146
/**
  * @see #openEditor(IEditorInput, String, boolean, int)
  */
private IEditorPart busyOpenEditor(IEditorInput input, String editorId, boolean activate,
		int matchFlags, IMemento editorState, boolean notify) throws PartInitException {	
    ...
    // line 3182
    setEditorAreaVisible(true);		
    ...
    if (desc.isInternal()) {
        // fine, the internal editor would need the "editors" area..
     } else if (desc.isOpenExternal()) {
        openExternalEditor((EditorDescriptor) desc, input);
        // no editor parts for external editors, return null
	return null;
    }
````

So RCP always shows the editors area, even for 'external' editors.

willrogers, 2015-11-05:
This is what I suspected.  As I said, our workflows don't seem to be using the navigator.  The time I find this is a problem is actually figuring out how to start the BOY examples, but that can be handled differently.

kasemir, 2016-04-14:
Reopening because this might be quite easy to fix in org.eclipse.ui.internal.WorkbenchPage: Just change
```
setEditorAreaVisible(true);     
```
into 
```
setEditorAreaVisible(desc.isInternal());     
```
Getting the fix merged into the Eclipse master will likely take longer, but still worth a try. The "Science" tools also tend to use views for plots etc. to display their data, so they might also be interested in opening views from the navigator w/o otherwise changing the part stacks.