ControlSystemStudio/cs-studio #1411: Opi runtime/view switch

berryma4, 2015-10-29:
pull request #1392

Because of current issues on master (4.2), and ITER needs to have something for a code freeze.  ITER has submitted a pull request to have the previous BOY behavior for their code freeze.  This is a fragment, so a person can choose.  I would rather we all move together on this core BOY change, but I also understand the pressures of having working code for a code freeze.

@kasemir, @shroffk, @willrogers, @utzeln   
comments on the pull request?

I would suggest, the need to have the new view behavior fragment as part of the boy feature, requiring someone to make their our feature if they want the old behavior.

willrogers, 2015-10-29:
This change makes the code more difficult to understand and maintain.

However, I acknowledge that right now there are effectively several regressions in BOY behaviour, or at least changes that would mean you have to change the way you use BOY.

willrogers, 2015-11-02:
Don't let me hold up the merge if you think this is the right thing to do.

kasemir, 2015-11-02:
The old behavior is on the 4.1 branch. I'm with Eric on rather moving forward on the master branch.

utzeln, 2015-11-03:
Does it mean that you are in favour of ITER forking its development branch?

kasemir, 2015-11-03:
We can talk about this in more detail tomorrow. In short, you may be better off looking at the ISIS approach that Matt C. described in a poster at ICALEPCS. They basically have a site-specific product that has certain fixed window sections, one line of buttons to select what you see, plus one section for *.opi files. That appeared pretty close to the ITER HMI guideline that you have previously shown, except their line of buttons is in the center, yours was at the bottom. They have a constant machine summary in the top-left, you wanted an alarm summary in the top right. So different pieces, but very similar overall idea to not use perspectives and movable parts.

berryma4, 2015-11-04:
We've agreed that the boy feature should include the view fragment. (so this is the default)
We've also agreed to help ITER move toward the view fragment by implementing missing requirements . @shroffk @kasemir  please check pull request

berryma4, 2015-11-05:
Jaka from pull request 

> This is already done. The opibuilder-features/org.csstudio.applications.opibuilder.feature from this 
> pull request contains the .view fragment. For anyone who is using that feature, no further actions are > required.
> 
> I realize that the names of the fragment plugins are maybe not the most adequate. I am open to 
> suggestions.
> -- <cite>Jaka Bobnar</cite>