ControlSystemStudio/cs-studio #1417: Managing third party libraries

shroffk, 2015-10-30:
Many of our plugins depend on third party libraries, after the new build system and repository cleanup there has been no clear idea as to how to manage these. There are currently two mechanism we can use.

**Old System:**
In this case we simply copy the jars into a new bundle manually created by cs-studio developers.

Advantages
- Cloning the git repository pulls in all the required binaries 

Disadvantages
 - The repository has binaries
 - Adding, removing, updating dependencies is more complicates, it requires manually adding new jars removing old ones, updating the export packages and the classpath.
 - Need a manually manage the dependencies (example https://github.com/ControlSystemStudio/cs-studio-thirdparty/tree/master/plugins/org.csstudio.platform.libs.jersey/lib)

**Maven-Osgi-bundles:**
In this method, developers added the newly needed artifacts to the following pom,
https://github.com/ControlSystemStudio/maven-osgi-bundles/blob/master/repository/pom.xml
After the first mvn install, tycho-maven will pull all the third party libraries/artifacts and create the bundles from it.

Advantages
 - No binaries in the repository
 - Adding, removing and updating dependecies requires updating the pom and rebuilding this repo
 - Maven can do the dependency management

Disadvantages
 - no documentation on how to use this
 - requires 2 networks calls (first to clone the repo, the second call is by maven which actually downloads the jars)...after which no further n/w calls should be necessary
 - while possible not sure how instruct maven to create bundles with the libraries untarred (for erformance reasons)

Note:
Both systems can be used simultaneous

Comments? @kasemir @willrogers @berryma4 @utzeln 
We could discuss this in more detail at the next meetins

willrogers, 2015-10-30:
I've tried keeping binaries in git repos and it's really expensive for the overall size of the repo.  cs-studio is very big already, so I'm generally in favour of avoiding this as far as possible.

kasemir, 2015-10-30:
I think we all agree that
* there are third party libraries (jdbc, pvAccess, jca, activemq, ..)
* we don't want them in the 'source' repository together with all the cs-studio *.java code. They're binaries, they don't change that often, they shouldn't slow down the development workflow that is mostly about the cs-studio *.java code.
* we do want a known target platform, no "it works for me" nor "what just happened?!"

Moving for example org.csstudio.platform.libs.jdbc and org.csstudio.platform.libs.jms out of the cs-studio repo into the cs-studio-thirdparty repo makes sense.
So far, org.csstudio.platform.libs.jdbc included the Oracle, MySQL, Postgres jar files. org.csstudio.platform.libs.jms included the Apache MQ jar files.
I'm unsure about the advantage of going much further.
Yes, we can remove those jar files, and instead put a pointer into a maven-osgi-bundles pom.xml so they're downloaded and wrapped. That means each time I do a complete rebuilt, they're downloaded. Maybe, depends on network access. Then there's the problem with the Oracle jar that we still need to keep as a copy since it's not on Orbit.
Finally, consider our proven inability to provide a target platform that actually works longer than a few weeks. The IDE target files that link to http:// repos simply don't work for me.
Those of us at the Melbourne meeting agreed that it's desirable to have a simple, reproducable target platform. Basically a target.zip that you download, unpack, and it includes all the *.jar files.
Some present mentioned that it's technically possible to provide that based on a zipped-up snapshot of the dependencies from the maven build. If that can be demonstrated, great. I've been struggling too much with maven/tycho, so personally I'd be fine with a "cs-studio-target" repo that's basically a binary repo: You clone it, and you have all the *.jar files. So our target becomes:
1) Specific version of Eclipse RCP/RAP IDE (4.5.1 for now)
2) clone of cs-studio-target

shroffk, 2015-10-30:
+1 for a zipped cs-studio-target p2 repository that serves as the simple target (since we have never been quite able to get a reliable target definition file to work)

However, the need to completely rebuild is something we should be avoiding...not resolving that issue with our build is partly the reason why our jenkins jobs are also failing.

berryma4, 2015-10-30:
When talking with Matt Gerring (Eclipse Dawnsci project) at the conference,
he mentioned they use a python script to create the target and simplify the
development setup.
(I know they use buckminster, but they are moving to tycho)
It would be interesting to get the details on this, see how simplified they
made it.

Also, when creating the thirdparty repos this was how I divided them:
cs-studio-thirdparty :  Thirdparty libraries that need to be modified in
some way
maven-osgi-bundles :  Thirdparty libraries that can be wrapped without
modification from maven central

kasemir, 2015-10-30:
One more example why I'd really prefer an un-zippable target platform instead of hoping for maven to download the missing pieces...

A few plugins, for example in diirt, need org.antlr.runtime. They don't depend on an antlr plugin. They only import-package org.antlr.runtime in the MANIFEST.MF. So it's up to every product which has to use diirt to also provide an antlr plugin that exports those org.antlr.runtime packages.

This can be tricky: Whoever assembles a product might know that they need diirt, but they don't know which version of antlr is required.

The common CSS product somehow ends up with org.antlr.runtime_3.4.0.jar, and that works OK.
The org.antlr.runtime_3.4.0.jar is actually in the diirt/p2/target/repository, maven-osgi-bundles/repository/target/repository, and the cs-studio/core/repository/target/repository

In the SNS product, I added org.antlr.runtime to the SNS core.feature so that it's included in my product.
Turns out maven ended up fetching org.antlr.runtime_4.3.0.v201502022030.jar from http://eclipse.mirror.rafal.ca/releases/mars/201506241002/plugins instead of using the org.antlr.runtime_3.4.0.jar from the diirt, maven-osgi-bundles or cs-studio/core repo.
Unfortunately, org.antlr.runtime_4.3 doesn't work, so diirt didn't load.

I've spent about 2 days looking at this error: org.antlr.runtime plugin is in the product, but diirt plugins complain that they lack org.antlr.runtime packages..
Finally, I saw the 3.4 vs. 4.3 version difference. I specifically add org.antlr.runtime version 3.4 to the SNS core.feature, and now it's OK.

willrogers, 2015-11-02:
I played this exact same game and ended up adding `org.antlr.runtime` version 3.4 to the DLS product.  I didn't know the 'correct' way to do this so I have just left it for now.

https://github.com/dls-controls/org.csstudio.dls.product/commit/7acd22c0ca5627f7f26e8fcc598395ca3615fd66

berryma4, 2015-12-30:
I added a pull request in diirt for this:
https://github.com/diirt/diirt/pull/65
@shroffk can we get a release for 4.2?
.... after someone else can verify the fix ( @willrogers, @kasemir )

kasemir, 2016-01-04:
Good, go ahead.

I still prefer `Require-Bundle:` over `Import-Package:`, because that way a simple dependency analysis gets you the list of required bundles, whereas import-package leaves it as an exercise to the student to figure out which plugin may provide the missing packages. But diirt doesn't have manifest.mf files. They're generated by the maven-bundle-plugin, which only offers `Import-Package:`, so this fix is the best that can be done.

shroffk, 2016-01-05:
Ok, So I tried the latest pull request submitted by @berryma4  with the version numbers defined as a range and the previous build issue is resolved.

berryma4, 2016-05-04:
What is the to do on this?

closed 2020-07-23 by shroffk