ControlSystemStudio/cs-studio #1424: Setting an invalid Diirt configuration path causes an unrecoverable error

nickbattam, 2015-11-03:
Setting the Diirt configuration to a "valid path", but one that doesn't contain the Diirt configuration XML files, e.g. /tmp, generates a large number of exceptions during the application startup:
FileNotFoundException, ExceptionInitializerError, NullPointerException
The exceptions are caught at a high level in the Eclipse framework which causes the startup to abort.

In the first instance I will extend the current checking in the preferences dialog to ensure the selected folder not only exists but also contains configuration files.

I will look also at the Diirt library itself to see if there's an 'easy' way to prevent an invalid preferences location property triggering a cascade failure.

kasemir, 2015-11-03:
Thank you very much for looking into this!

The preference dialog is certainly a nice addition, but I'd put emphasis on the overall application behavior. Most users will interface with a pre-installed or downloaded CSS setup that's already preconfigured. Such an installation may log error messages, but it should not abort on startup.

Even those who use the preference GUI may select a valid location in there, but over time that location is renamed/removed, and again CSS should then log errors, show disconnected PVs, but not abort.

So yes, please, look at the Diirt library itself as necessary to turn such errors into messages instead of the application abort.

nickbattam, 2015-11-04:
I've been unable to reproduce the original issue. 

As @shroffk suggested, if the config path is invalid or doesn't contain configuration files a default (no CA) configuration is created which stops catastrophic failures.
There as still many NullPointerExceptions raised but these are harmless.

The only possible failure point is in the Diirt preference page which didn't handle null data in the treetable well.

shroffk, 2015-11-10:
Can we close this?

nickbattam, 2015-11-10:
Yes

closed 2015-11-10 by berryma4