ControlSystemStudio/cs-studio #1440: Exception from JCAPV.asyncRead()

willrogers, 2015-11-11:
I'm trying to use `org.csstudio.vtype.pv`.  I get an exception from the following test (which I wrote for the purpose in `JCAPVTest.java`):

```java
    @Test
    public void testAsyncRead() throws Exception {
        final PV pv = PVPool.getPV("motor_x");
        Future<VType> pvFuture = pv.asyncRead();
    }
```

Here is the exception:

```
java.lang.IllegalStateException: Channel not in connected or disconnected state, state = 'NEVER_CONNECTED'.
	at com.cosylab.epics.caj.CAJChannel.checkState(CAJChannel.java:1090)
	at com.cosylab.epics.caj.CAJChannel.getFieldType(CAJChannel.java:732)
	at org.csstudio.vtype.pv.jca.JCA_PV.asyncRead(JCA_PV.java:250)
	at org.csstudio.trends.databrowser2.command.NewWindowHandler$DatabrowserOpener.run(NewWindowHandler.java:58)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:35)
```

* Is this the right PV library to use?
* Am I doing something wrong?

kasemir, 2015-11-11:
The channel is still in the state of getting connected.

Compare with the examples in org.csstudio.vtype.pv.test/src/org/csstudio/vtype/pv/JCAPVDemo.javaw which all wait for the first value that indicates that the connection & meta-data-read completed.

This is a side effect of the direct, non-decoupled behavior of this lower PV layer, as compared to the PVManager. I've thought about adding something like "boolean isConnected()" or "Future<> awaitConnection()" to make writing tests a little easier:
''''
final PV pv = PVPool.getPV("motor_x");
pv.awaitConnection().get();
// XXX
Future<VType> pvFuture = pv.asyncRead();
''''

but that doesn't really help because your PV can get disconnected at time XXX.

So for now you get the exception from the underlying JCA/CAJ when you try to read/write while the channel is not-yet connected or became disconnected.

willrogers, 2015-11-11:
The way I naively expected it to work:

```java
    final PV pv = PVPool.getPV("motor_x");
    Future<VType> pvFuture = pv.asyncRead();
    // this will block until ready or timeout
    pvFuture.get()  // or pvFuture.get(5, TimeUnit.SECONDS);
```

It could very well be that this is difficult or impossible using JCA, I certainly haven't dug very far into the implementation.  Shall I just close this ticket?

kasemir, 2015-11-11:
I can argue both ways:

A) `asyncRead()` issues a read request (for CA a get-callback) and awaits the value.
If it can't issue the read request in the first place, because for example the PV is disconnected, that's an error.

B) `asyncRead()` asserts that the PV is connected. If not, it waits until the PV is connected. Then it issues a read request (for CA a get-callback) and awaits the value.

The former is the current implementation.
The latter appears reasonable, but then what about the asyncWrite?

A) Issue a write-callback, wait for  the completion. Disconnected, no write access are all errors.
B) Wait for connection. If no write access, wait for write access to become available? Then issue write-callback, await completion.

The wait-for-write access makes sense when you really want to wait until it all happens, whatever it takes. But I think it goes too far.

If we stay with the simpler A cases, maybe utility functions to check connection state and await connection should be added to simplify writing tests?
For applications that care less about the current state of the PV and direct updates but prefer decoupled updates, the PVManager is supposed to provide that.