ControlSystemStudio/cs-studio #1446: products repo not built for 4.2?

MichaelRitzert, 2015-11-17:
I'm trying to test my code against the 4.2 code base of CSS. However, the maven build fails with
    [ERROR]   Missing requirement: css-b2vxd 4.2.0 requires 'org.csstudio.product.feature.feature.group 0.0.0' but it could not be found
while debugging this, I find that the product repo is mentioned in
    http://download.controlsystemstudio.org/updates/4.2/compositeContent.xml
but in a strange way:
    <children size="4">
        <child location="http://download.controlsystemstudio.org/applications/4.2"/>
        <child location="http://download.controlsystemstudio.org/maven-osgi-bundles/4.2"/>
        <child location="http://download.controlsystemstudio.org/thirdparty/4.2"/>
        <child location="http://download.controlsystemstudio.org/core/4.2"/>
        <child location="http://download.controlsystemstudio.org/product/4.2"/>
    </children>
Note the "size=4" with 5 entries following. Probably because of this, maven doesn't try to download from the repo.

However, when I manually add the location to my pom.xml, I cannot find a repository there. I don't know enough about maven to judge if this is related or a separate issue.

berryma4, 2015-11-17:
Yes, this is waiting on:
https://github.com/ControlSystemStudio/cs-studio/issues/1356

berryma4, 2015-12-22:
completed:
http://download.controlsystemstudio.org/release/4.2/

closed 2015-12-22 by berryma4