ControlSystemStudio/cs-studio #1456: BOY Example 1_2_WidgetExamples.opi does not render

berryma4, 2015-11-18:
runtime:
[image]

editor:
[image]

shroffk, 2015-11-23:
Was there an exception when you opened this page?

In the nsls2 development product, the page takes a significant amount to time to open (10s of seconds). However the page does successfully open and dispplays most widgets

The problem I see is the dotted uninitialized pv errors persist on the boolean symbol widget even after a value is set to them

[image]

berryma4, 2015-12-01:
After fixing features, and removing pvmanager 2.x references #1470 this has gone away

closed 2015-12-01 by berryma4