ControlSystemStudio/cs-studio #1460: Accessing Beast Alarm information in CS-Studio applications

utzeln, 2015-11-19:
Goal: To able able to display alarm information from the beast server and be able to write basic messages.

shroffk, 2015-11-24:
There are various ways of accessing this information. Currently the information is split between 2 locations 
1. the jms messages
2. the configuration database

I have added a new prototype jms datasource in diirt which can be used to listen to and send jms messages. The datatype has TypeAdapterSet which can be expanded to add new adapters for converting different jms messges to Vtype.
https://github.com/ControlSystemStudio/diirt/tree/jms_datasource

However the jms datasource alone is not enough since the jms messages only contain the live alarm information.

So a new "beast" datasource was added, this datasource uses the AlarmClientModel provide in the o.c.alarm.beast plugin.

It can be used as follows 
beast://SR-RF{Xmtr:D-Cav}Pwr:Fwd-I
Currently the return type is a VTable with 2 columns (both VString) which contain the useful bits 

[image]

The current datasource allows you to write strings to it and then translates those strings to appropriate actions
ack:       acknowledges the alarm
unack:   unacknowledged an alarm
enable:  enables a pv
disable: disables a alarm pv 

TODO:
The current topics which need discussion are
1. How to effectively support multiple alarm config trees and managing the initialization of the beast datasource.
2. How to connect to nodes in the alarm tree.
3. What are the return types that need to be supported.
4. What should be the syntax of the beast channels.

shroffk, 2015-11-25:
Implemented a prototype for registering to nodes.

The new syntax for the datasource is as follows
beast://complete/path/to/node
beast://pvName

[image]

boris-v, 2015-11-26:
I have made some progress with the widgets' alarm behaviour being driven by the BeastDataSource (only tested with TextInput so far).

A couple comments I have:
1) Right now the AlarmState received from the BeastDataSource is the latched severity.
I'd prefer to default to the current alarm sensitivity behaviour, where the border/background changes based on the current alarm severity.

Also, given the AlarmTable behaviour requested by ITER, I believe we will also need the current severity in the not-too-distant future to mimic the "cleared unacknowledged alarm" visual feedback.
@utzeln Would I be correct with such an assumption ?

@shroffk:
2) Acknowledged right now is the same as AlarmStatus (BeastMessagePayload.isAcknowledged returns the severity). I think we could get rid of it in this form ?
I've recreated BEAST's SeverityLevel in opibuilder.util (at least for now - should I rather delete it and go back to if-elses?), so we'll be able to use the same Acknowledged logic as the AlarmTable.

3) I had some weird behaviour during testing: my AlarmPV-driven sensitivity only worked after I was in the Alarm perspective at least once.
When starting CSS in the Editor, it didn't work. (See screengrab.)

[image]

The problem is that the BeastDataSource crashes in the findPV method because it is called before the model is present (set by the async initialization in the constructor).
It crashed when I created my listener, inside the createChannel method where it calls findPV. I'm not sure yet if this happens because I did something wrong or not (maybe initialize it early ?).
You can see how I currently create the PVReader here:
https://github.com/css-iter/cs-studio/blob/a35c771ad66b36a7e6b5011ac47669e05f064f28/applications/opibuilder/opibuilder-plugins/org.csstudio.opibuilder/src/org/csstudio/opibuilder/editparts/PVWidgetEditpartDelegate.java#L247

I do have a temporary workaround in place for now: in the beginning of createChannel I sleep for 1 sec, this gives the initialization enough time to complete with my small alarm tree.
As for a solution, I thought about waiting for the model in createChannel and creating the listener with async, but I'd prefer to be able to create the listener even if the datasource isn't initialized yet, and only start receiving events after it's completed the init stage.
What do you think ? (Unless I did miss a step somewhere..)

boris-v, 2015-11-27:
Using a TextInput widget, alarm-sensitive behaviour now works using a separately set beast:// 'AlarmPV', with rudimentary support for ack/unack from the context menu.
Right now it uses the current alarm severity, not latched, for ease of testing.

@shroffk Additional comments:
For now I've settled with a blocking wait on createChannel for ease of testing. This was the only way I found that I could test both with a full path PV and a name-only PV, without recreating readers or starting from the Alarm perspective.

As far as I can tell, without it I have some trouble either way because of the key for the monitors map in diirt & key for the BeastDS' internal map:
- if the channel is created before the model exists, createChannel will throw a NullPointerException and the channel will not be created -> my widget listener won't receive updates
- if I modified BeastDataSource to create the channel without trying to resolve the PV and create the channel via full path, the channel was created and the monitor bound to the PV name; however, even if I close & recreate my reader after this point I still won't get updates
- if I modified BeastDataSource's initialization to recreate the channels for me after it receives the model, I still didn't get updates (I think because at this point the diirt 'registered' reader was mapped to the PV name but the internal beast map was to the full path)

The last two only happen when I use just a PV name in the widget, like 'beast://TEST-BOY0:CMDSWITCH'; if I use the full path (/CODAC_AlarmHandler/TEST/TEST-BOY0:CMDSWITCH) it worked fine after recreating the listener.

shroffk, 2015-11-27:
1. The need to switch to the alarm perspective can be avoided by setting a default config root in the beast.xml configuration file.

2. I have added current severity to the beast message payload

3. Regarding the creation of the channels, this should work even when the model is still being loaded...the datasource will create the channelhandlers and then trigger a reconnect when the alarm model has completed loading. In the meanwhile the PV's should remain in the "connecting" state. I see this behaviour in the probe but I might be missing some use cases so we can go over it in detail...and I can fix the datasource, you should not need a sleep in your code.
In anycase, with the new node support, I plan to deprecate the findPv method and instead use the getState method which takes care of handling both nodes and leaf.

boris-v, 2015-11-27:
Ok, I'll merge the changes in with mine (the VTableAdapter; I've added CurrentAlarmStatus too, removed ack. and renamed AlarmStatus to LatchedAlarmStatus).

The reconnect wasn't enough in my testing (CSS started with OPI Editor perspective, then run OPI with a beast AlarmPV -> initialization takes a bit longer in this case): the channelhandlers are created and mapped to channel (e.g.) "TEST-BOY0:CMDSWITCH" (but I had to add a null check to findPV), but after the Alarm tree model is available this is expanded to the full node path (e.g. /CODAC_AlarmHandler/TEST/TEST-BOY0:CMDSWITCH), and the latter is the key in the map for the new handlers. At least that's how it behaved when I debugged it O_O

IIRC my PVReader was bound to the short channel name in the 'monitors' map in PVDirector (or one of those classes, can't remember at the moment) and this might have bugged it. I'll check through again and write it down, I tried several different ways around this and I'm afraid I might be mixing up the details.

boris-v, 2015-12-12:
@shroffk 
The current BeastDataSource still doesn't propagate received events when a channel is created with only the PV Name, e.g.:
`PVManager.readAndWrite(formula("beast://TEST_1-0"))...`
In this case the `BeastDataSource.map` entry will have the key "TEST_1-0".

The BeastDataSource's `AlarmClientModelListener.newAlarmState` only "tries" to notify PVReaders that registered to the full path:
`List<Consumer> handlers = map.get(pv.getPathName());`
In my example, `pv.getPathName` returns my `AlarmTreeItem`'s full path: 
"/CODAC_AlarmHandler/TEST/TEST_1/TEST_1-0", so `handlers` is `null`.

Thus, the reader from the example above will not receive any events after the initial one.

A naive fix was duplicating the handlers execution code for those registered with the pv-name :
(https://github.com/shroffk/cs-studio/blob/Beast_dastasrouce/applications/alarm/alarm-plugins/org.csstudio.alarm.diirt.datasource/src/org/csstudio/alarm/diirt/datasource/BeastDataSource.java#L109)

```
// listeners that registered with only the PVName as the channelName instead of the full path
handlers = map.get(pv.getName());
if (handlers != null) {
    for (Consumer consumer : handlers) {
        consumer.accept(pv);
    }
}
```
This covered both PVName- and PVPath-registered channel readers of the same PV and is what I'm currently using for testing.

shroffk, 2015-12-14:
I see this, I am looking into fixing this

shroffk, 2015-12-14:
I have fixed the problem
If you create a the following channels
beast://pvName 
beast://complete/path/to/pvName 
Both which be managed as separate beast channels however when there is a newAlarmState, both of channels will be notified on the change

Additionally I also added the fixes to ensure that when there is a alarm state change, all parents are also notified.

Finally I have added support to enable/disable(AlarmTreeItem) nodes in addition to pvs (alarmTreePVs).

shroffk, 2015-12-14:
The unnecessary acknowledge field has also been removed from the table adapter.

kasemir, 2015-12-16:
Regarding today's hangout with the proposal to have all opibuilder widgets become aware of this alarm data source, so any widget with a PV 'X' would also try to connect to 'beast://X' and indicate the alarm state in some way, allow acknowledge etc:

This should not be done in a way where the opibuilder depends on the alarm plugins. We cannot require every product that uses a display to also include the alarm client code. A configuration setting that just disables the alarm client functionality is not sufficient. This needs to remain modular such that the basic opibuilder and the basic alarm plugins have no connection. A new plugin o.c.opibuilder.widgets.alarm that has one or more alarm-aware widgets would be OK, but having the basic o.c.opibuilder plugin depend on the alarm plugins is no good idea.

jbobnar, 2015-12-17:
@boris-v 
Found a great way how to do this (almost) without touching the opibuilder. See here: https://wiki.eclipse.org/Steps_to_use_Fragments_to_patch_a_plug-in
All that needs to be changed in the opibuilder are a few lines in build.properties and manifest.mf files. I just tried it out and it works like a charm. Than every lab can decide whether to include the patch fragment (and have the beast functionality) or not (the current behaviour).

@kasemir  
In any case even the patch fragment will not have any dependency on the alarm stuff. It will only depend on diirt (simplepv actually). Diirt is responsible to handle all the alarm stuff.

shroffk, 2015-12-17:
I agree with Kay that the alarm dependencies creeping into core BOY is not a good idea. So if we can manage it with fragments I would vote for that.

If we manage to get the fragment setup working, like Jaka mentioned there might not be any direct dependency to the beast plugins. However opibuilder would have to be smart enough to realize when the beast datasource is and isn't available...The simplest way might be to have a preference in addition to the fragment which would decide if we should skip the creation of beast channels completely.

@boris-v: I am still working on the timestamp

kasemir, 2015-12-17:
Is the alarm data identified in any way, or is it just a table that happens to contain alarm info, "duck typing"?

For what it's worth, the NTTable allows for optional descriptor and alarm:
```
NTTable := 
structure
    string[]   labels              // Very short text describing each field below, i.e. column labels
    structure  value
        {scalar_t[]  colname}0+ // 0 or more scalar array instances, the column values.
    string     descriptor  : opt
    alarm_t    alarm       : opt
    time_t     timeStamp   : opt
```
Maybe the descriptor could be used to identify alarm info, so special widgets would recognize it as such.
The alarm info could be used to allow even generic table-aware widgets to be alarm-sensitive.

boris-v, 2016-01-04:
@shroffk It might be worthwhile to have the BeastDataSource release its reference to the AlarmClientModel a set time after there are no more readers for it, and acquire it again when needed (e.g. if no new readers connect within 30 seconds).

Not just to free resources, but also because an AlarmClientModel instance for a given configuration is a singleton.

During testing with simulated network problems (CSS couldn't access server at varying points), I noticed that if I start an OPI that uses the BeastDS while the server is unreachable (AlarmClientModel can't init), even if the server becomes reachable again I am forced to restart CSS, because the single instance for the given configuration will stay `alive` and disconnected while CSS is running (diirt / BeastDS keeps a reference).

Any new PVReaders created after this get the org.diirt.datasource.TimeoutException exception.
This doesn't impact just the Beast functionality in OPIs, but also any OPI that has an AlarmTableWidget for the same config, as it will receive this same instance that diirt is holding on to.

I can force a reconnect of the Alarm perspective by closing and reopening the perspective, but can do nothing for OPIs.

** If BeastDS' model is instantiated (without problems) and the network goes down *after* this point, it will reconnect normally when the server is reachable again (while the OPI is running).

If releasing the model when it's not needed is a no-no, what about releasing and retrying if it failed during BeastDS' initialization ?

shroffk, 2016-01-05:
@kasemir "duck typing" was the easiest way to move ahead while we figured the architecture out.

Currently the VTable does not have fields for a collective table wide alarm and timestamp

Although I would like to return a VTable with columns whoes types are more accurately set to represent the data in it.

i.e.

Name (VString) | Value (VType) | AlarmState (VAlarm) | Active (VBoolean)  
--------------------|--------------------|----------------------------|------------------------
vacuum | 1.0 | Major-ack | true 

So, the value from from columns can be directly used in most cases.

shroffk, 2016-01-20:
Channels connection states are now updated with server connection/disconnection events.
Reconnection is automatically attempted on recovery

boris-v, 2016-01-20:
@shroffk 
I had to tweak BeastDataSource a bit so that leading/trailing slashes work correctly:
/demo/test-node/   /demo/test-node   demo/test-node/   demo/test-node

When retrieving the initial state from the model the BeastDS correctly got the same node regardless of "/", but when dispatching events it only sent it to readers mapped to (e.g.) "demo/test-node" (parentHandlers in code).

For this I've overridden the channelHandlerLookupName (for diirt's usage) and used it in add() & remove():
https://github.com/css-iter/cs-studio/blob/master/applications/alarm/alarm-plugins/org.csstudio.alarm.diirt.datasource/src/org/csstudio/alarm/diirt/datasource/BeastDataSource.java#L168

Do you want a PR, not having such code (require root/node channel names), or will you implement it differently ?

shroffk, 2016-01-20:
Ok, I am fine with this change

slight changes
```
        if (channel != null && !channel.equals("/") && !channel.isEmpty()) {
            if (channel.endsWith("/"))
                channel = channel.substring(0, channel.length() - 1);
            if (channel.startsWith("/"))
                channel = channel.substring(1);
        }
```

I will looking into checking if there is away to handle the channelname parsing with the URI class.

boris-v, 2016-01-20:
I had it written with *With(), was suggested to change to charAt checking for single-char :)

shroffk, 2016-04-25:
#1748