ControlSystemStudio/cs-studio #1485: "Name of user" lower rectangle too small to read username

berryma4, 2015-12-01:
Lower left corner:
[image]

berryma4, 2015-12-02:
@kasemir

kasemir, 2015-12-07:
Is that on windows? Or on a newer Linux setup?

berryma4, 2015-12-07:
Windows 7, I believe Dylan it's setting our on Linux.

berryma4, 2015-12-07:
@dylan171, which version of Linux do you see this on?

berryma4, 2015-12-08:
https://bugs.eclipse.org/bugs/show_bug.cgi?id=471313

berryma4, 2015-12-08:
@kasemir I've submitted a workaround.  I had to add a label, so now a "-" precedes the logged in box.
Are you okay with this, until the eclipse issue is fixed?

kasemir, 2015-12-08:
I'm OK with that. Maybe add
```
<!-- Workaround for https://bugs.eclipse.org/bugs/show_bug.cgi?id=471313 -->
```
to the plugin.xml change.

berryma4, 2015-12-08:
I added the note.
Thank you!

closed 2015-12-08 by berryma4

kasemir, 2015-12-08:
Thanks for fixing this before I've ever noticed that there's a problem!