ControlSystemStudio/cs-studio #1490: the default property not applied for: org.csstudio.utility.product/cs_studio_perspective

berryma4, 2015-12-01:
I had to set this property in the product plugin_customization.ini for it to work.
The default seems to be ignored, or overridden.
(testing with the org.csstudio.product common product)

org.csstudio.utility.product/cs_studio_perspective=org.csstudio.diag.pvmanager.probe:org.csstudio.diag.pvmanager.probe:left:multiple;\
 org.csstudio.diag.probe:org.csstudio.diag.probe.Probe:left:multiple;\
 org.csstudio.utility.clock:org.csstudio.utility.clock.ClockView::;\
 org.csstudio.trends.databrowser2:org.csstudio.trends.databrowser.archiveview.ArchiveView::;\
 org.csstudio.alarm.beast:org.csstudio.alarm.beast.ui.alarmtree.View::;\
 org.csstudio.alarm.beast:org.csstudio.alarm.beast.ui.alarmtable.view::;\
 org.csstudio.diag.pvutil:org.csstudio.diag.pvutil.view.PVUtilView:left:;\
 org.csstudio.opibuilder:org.csstudio.opibuilder.opiView:center:multiple;

berryma4, 2015-12-02:
@shroffk

shroffk, 2015-12-02:
I have had trouble getting the product customization to work correctly.
Working with the win32, x86_64 version of the cs-studio common product, built of the master branch of the org.csstudio.product repo. 
The plugin_customization.ini file while present in the product is not being used to apply the defaults.

The recent changes in terms of removing of the org.eclipse.rcp feature might be one possible explanation.

closed 2015-12-04 by berryma4