ControlSystemStudio/cs-studio #1494: Option not to show readout on Spinner widget

willrogers, 2015-12-03:
When converting an Up/Down button from EDM, there is no corresponding widget.  What it really requires is a Spinner widget without a readout label  There are two options:

* Make it possible not to show readout on the Spinner readout.  This would allow correct conversion
* Hard-code the conversion to show only the button part of the Spinner widget.  This is done in https://github.com/ControlSystemStudio/cs-studio/pull/1493

I'm happy to do whichever is agreed on.

willrogers, 2015-12-08:
Hold on there's still a question here.

Should the Spinner widget have an option not to show the readout?

kasemir, 2015-12-08:
Yes, I think that would be the proper thing to do, since your current way of hiding the readout depends on knowing the width of the up/down button part. Also assumes that the up/down buttons are on a specific side of the widget. In an updated version of the UI, such a narrow widget width could result in only showing the first few digits of the readout, hiding the _buttons_.

willrogers, 2015-12-08:
Exactly - this is a workaround because the widget doesn't have the feature we want.  We will work on this.