ControlSystemStudio/cs-studio #1495: .opi files have mixed line endings

tboegi, 2015-12-04:
It seams as if CSS stores opi files with "mixed line endings":
The first line is terminated with a line feed (LF or "\n"):
<?xml version="1.0" encoding="UTF-8"?>

The next lines are terminated with carriage return line feed (CRLF, "\r\n"):
<display typeId="org.csstudio.opibuilder.Display" version="1.0.0">^M

And the last line has LF.

Is there a reason to have CRLF in some lines ?

willrogers, 2015-12-04:
This seems to be true and is indeed very strange.  There is no reason for this behaviour.

I think we should write Unix-style line endings.

berryma4, 2015-12-04:
I agree!  Can we make the elements alphabetized while we are at it? (So
diffs on opis aren't as crazy as they are)

closed 2015-12-04 by berryma4