ControlSystemStudio/cs-studio #1499: Perspective handling: Command line option to install workspace.xmi

kasemir, 2015-12-07:
Since BOY runtimes execute in a "View", Perspectives can be used to define several display layouts.
For example, control room operators or beam line administrators can create several perspectives, and users can then switch between them.

Right now, however, there's no good way to add such predefined perspectives to a newly started CS-Studio instance.

Ideal would be a way to automatically import selected perspectives. For example, desired perspectives could be exported, one file per perspective, into some directory, and a startup option "-import_perspectives /some/dir/*" would import them all. Unfortunately, this is currently difficult because the original Eclipse perspective import/export mechanism is broken. There is progress on the issue, https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811, so waiting for that to be fixed may be a good approach for us to take.

Another option is to replace the `.metadata/.plugins/org.eclipse.e4.workbench/workbench.xmi` file on CS-Studio startup with a ''golden' `workbench.xmi` file that has been prepared to contain all the desired perspectives. In addition, this approach will assert that CS-Studio always starts up with a known initial display layout. On restart, user adjustments are lost and CS-Studio again opens with the 'golden' display layout. For some beam line use cases, this is actually desired, so it remains a useful option even after the selective perspective export/import becomes available.

berryma4, 2015-12-08:
related to #1398 

and,
https://bugs.eclipse.org/bugs/show_bug.cgi?id=378811
Verified in the Eclipse integration build I20151207-2000

closed 2015-12-08 by berryma4

willrogers, 2015-12-10:
This works surprisingly well.  Thanks!

kasemir, 2015-12-10:
It does a complete replacement of the workbench.xmi, so on restart you lose which windows you had open etc. For beamlines, our current plan is a "CSS" desktop icon for end users that will start css with the new `-workbench_xmi /path/to/common/workbench.xmi` option. People who start css on the command line can decide if they want to add the option; by default, nothing changes for them.

PR #1500 (2015-12-08): Workbench 1499
Adds "-workbench_xmi" command line option for installing a workbench.xmi file into the workspace.
Can be used to assert known initial display layout and predefined perspectives.
Fixes #1499