ControlSystemStudio/cs-studio #1515: PVTable Updates

kasemir, 2015-12-11:
From Charles-Henry PATARD:
[image]

We finally finished our PVtable coding. We’re ready to propose you new PVtable functionalities. So, just below a short overview :
-        Table export to Excel file
-        New column Timestamp Save which save the timestamp for each snapshot
-        Ability to create a configuration list of PVs if a line start with #conf#
-        If a configuration list exists in the table the user can  “make a measurement”. Make a measurement is the ability to take a snapshot of the configuration PV list and add the snapshot at the end of the table.
-        The user can delete the last measure or all the measures.
The native behaviors of PVtable still available.
 
We committed on master branch of cs-studio git repo. We would like to know if those features are good and useful for cs-studio project ?
So, now I would like to know how to create a pull request ?

kasemir, 2015-12-11:
You describe several features.

Table export to Excel is good, we should merge that.

New column TimestampSave that holds the time stamp of the saved value is good. Except for maybe labelling it "Saved Value Timestamp", I'm for merging that. In principle, the visibility of that column could be configurable just like the "show_description" and "show_units" columns which were added but can be enabled/disabled via preferences. I'm leaning towards always showing it, though, since the time when a value was saved should be generally useful.

As for the configuration list of PVs, I'm afraid I don't understand how that works and what it's good for.
Showing the same PV several times within the same table, with different "measurement" values, clashes with the original design of the table which takes a snapshot of PVs and then allows you to restore all of them, or selected ones. Restoring all of them would be very confusing if the same PV is listed multiple times with different values.

kasemir, 2015-12-11:
Forgot to answer your question on how to create a pull request.
We use the basic github workflow, see for example https://help.github.com/articles/using-pull-requests/,  https://help.github.com/categories/collaborating/.

Basically:
1) You "fork" the repository via the "fork" button on the github page.
2) You can then fetch a clone of that fork, change the sources in there.
    You could do that on "master", but typically we use some branch like "issue_1515" that includes the ticket number.
3) When done, create a pull request from your_fork:issue_1515 into cs-studio:master

If the pull requests come in small pieces like "Excel table export" where we've already agreed on the ticket that it's a good idea, it's quickly merged.

berryma4, 2015-12-11:
some of this overlaps with save restore:
https://github.com/ControlSystemStudio/cs-studio/issues/1516

I love the pvtable for what it is.  I don't know about extending it to add measurements/snapshots.  How does this work restoring-wise?  I think I might need to see it demo'ed.

berryma4, 2015-12-11:
I like that the measurements/snapshots are saved in a queue-like, I not sure about it visually though.  Like I mentioned, maybe I just need to see a demo.

kasemir, 2015-12-11:
Indeed I see the PVTable and save/restore as separate, but related.
Save/restore creates several snapshots.
It might allow opening one such snapshot as a PV table, so in the table you can then compare the saved value from the snapshot with live data, adjust the live value, restore all or selected PVs from the snapshot.
Still means each PV table contains just one snapshot per PV.
A series of measurements would turn into several snapshots, i.e. that would be done within the save/restore tool.

Should we have a hangout on this?

berryma4, 2015-12-11:
yes, that sounds good.  Let's have a hangout.  I can schedule it.

charlesHenryPATARD, 2015-12-14:
Hello,
I followed your advises. I created a boolean preference show_saved_timestamp and the column is hide or show. Excel export doesn't export Description and Saved value Timestamp if preferences are setted.

I created a demo video of Configuration and Measurement behaviours. I will post it on this page tonight. The needs of our users is to be able to take multiple snapshot of list of PVs at different values and time. Like this, they don't have to copy and past many times a PV list and take a snapshot. When you create a measure from a configuration list a snapshot is take and a copy of snaphot values is used to create the measure.

If you don't create a configuration list by typing #conf# PVtable operate normally.

charlesHenryPATARD, 2015-12-14:
If you want to have a look to demo : https://drive.google.com/folderview?id=0Bzw-SHJzUmojX3p5UnJOQ1lGQ0k&usp=sharing

kasemir, 2015-12-14:
Looking at the video, what you call Measurement is what I'd call Save/Restore.

Save/Restore is meant to save more than one snapshot of values for a list of PVs, and later you can then see what the values were at some time, compare, restore.

__One__ such snapshot could be presented in a PV table, where you can then compare, restore all values, or only restore values for selected PVs.
Storing more than one snapshot in the same table doesn't make sense to me because then the "restore all" is not defined. You would write several different measurements to the same PV.

kasemir, 2015-12-14:
Can we have a hangout on this Dec. 15, 10:00 eastern time/16:00 France?
Or Tursday, 8:00 eastern time/14:00 France?

charlesHenryPATARD, 2015-12-15:
Yes of course, tomorrow at 2 p.m. Can we have google hangouts  for discussing ? Have you got a google account ?

kasemir, 2015-12-15:
You mean tomorrow, Dec, 15, 08:00 eastern, 14:00 France?
Eric, can you make that, since the integration with resp. distinction from save/restore is kind of key here?
Or tomorrow, Dec, 15, 10:00 eastern, 16:00 France, which would be after the alarm hangout https://plus.google.com/u/0/b/101349549663920375487/events/cvcp67pr7ed46rjfm3irspt4bt0?authkey=CLTW_vO2uoO9FQ?

charlesHenryPATARD, 2015-12-15:
I mean Thursday, Dec. 17, 8:00 eastern time/14:00 France. I would like to show you my screen. So we're trying to create a call conference meeting for Thursday.

kasemir, 2015-12-15:
Created hangout event here: https://plus.google.com/u/0/b/101349549663920375487/events/c2p0enb9vs9mchtm9defqr3ntbs?authkey=CImY68rxmN2wRA

kasemir, 2015-12-17:
Notes from hangout w/ Charles-Henri, Eric, Kay:

Eric presented save/restore tool.
Pluggable way to select configuration (example: periodic table).
Configs can take several snapshots, one is then saved w/ description.
Saved snapshots can be opened, compared, restored.
Backend is git.

Charles-Henri presented PVTable update, (see video in ticket).

Conclusion:
* Merge “Saved Value Timestamp” column
* Merge Export to excel
* Merge messages_FR.properties
* Merge measurement support after a few adjustments:
  - Check spelling of XML file tags
  - Hide measurement toolbar buttons unless there is a “#conf#” entry
  - Update online help

Save/restore might support opening one saved snapshot as PVTable.

anthophil, 2016-01-20:
Hello everyone,

I'm working on the next PVTable update to add behaviors that you wish kasemir. But i'm encoutenring a problem to hide measurement toolbar buttons. I have changed the way i've defined the measurement toolbar because of a difficulty to obtain a good behavior to hide/unhide it. Now, the toolbar is define in the plugin.xml file and it works. Unfortunately, i can't place it after the tolerance button.

I'm short of idea. I've tried to set the toolbar location uri after the PVTableActionContributor but it doesn't apear at the right location. I've tried to set the location uri at the end of the toolbar but it apear at the right extremity, after perspectives, so this is not satisfying. I've tried to set it after the search button but whatever i try, it apear at the same place, after the Probe button.

Any idea ?

kasemir, 2016-01-20:
No idea. I've certainly seen that the location of toolbar sections contributed by different sources moves around as perspectives or versions of Eclipse change. The only guarantee seems to be that the buttons within one section stay in the same order.

So what you might have to do is: Contribute all PVTable tool bar buttons programmatically from the PVTableEditorActionBarContributor:
````
public class PVTableEditorActionBarContributor extends
        EditorActionBarContributor
{
 ...

    @Override
    public void contributeToToolBar(final IToolBarManager mgr)
    {   // Original buttons that are always included
        mgr.add(snap);
        mgr.add(restore);
        mgr.add(tolerance);
        // Optional buttons
        if (show_those_other_buttons)
        {
            mgr.add(save_measurement);
        }
    }
...
````

laurent-PHILIPPE, 2016-01-20:
Hi Key,

We have trouble with the programmatically method for the following reason.

By example, if you start with a new PV editor, the Editor doesn't have a config so the measures actions are hidden.
When you  add a config, we should add the measures action but we don't pass through the contributeToToolBar method at this moment.

Consequentally we add the action into IToolbarmanager and force the update in the model changed event of the PVTableEditor. But the result is not always satisfactory, sometines eclipse doesn't refresh the toolbar correctly.

 We will do another try tomorrow with anthony, but if we don't succeed, Are you agreed if we define all PVTable toolbar action with the menu extension point/command method (and the visibleWhen condition) in the pluggin.xml?

kasemir, 2016-01-20:
As long as you can hide the measurement buttons for normal PVTable configs that don't use the “#conf#” entry, I don't care how yo accomplish that.

charlesHenryPATARD, 2016-01-29:
Hi,
We're ready to push the last version of PV Table. We commited changes to master branch in our working directory. So now I would like to push changes to cs-studio remote. I can't create a Pull request by git hub because I can't find the issue 1515 in compare branches.
I can't push to master branch on https://github.com/ControlSystemStudio/cs-studio.git because access is forbidden (I think it's logical, I haven't rights to push on cs-studio master).
Can you help me please ?

Thank you

berryma4, 2016-01-29:
A pull request from a fork is the easiest way to let others test and okay the changes.
Thank you!

berryma4, 2016-03-07:
We have a few compile problems: https://openepics.ci.cloudbees.com/job/cs-studio-applications-display-master/44
Also, it might be nice to fix the tabs (check style errors)

charlesHenryPATARD, 2016-03-07:
I will work on it tomorrow morning. Can you explain me exactly what you need and what's the problem @berryma4 ? I never used Jenkins at least. I understand that there is a problem with the pom file and with unit test on Pace tool. But I nerver modify any java code in org.csstudio.display.pace.test.
It's difficult to answer to all recommended practices by cs-studio developers community. It will be nice if we have all good practices and tools details in a document. The name can be  "best practices for developers" :

- How to get an updated Target Platform
- How to use meaven and artefacts 
- How to use Jenkins

Thanks

willrogers, 2016-03-07:
I suggest it should probably be on the wiki: https://github.com/ControlSystemStudio/cs-studio/wiki

Also to add:

* How to set up unit tests
* Code style guidelines

berryma4, 2016-03-09:
With the last commits, does everything work "as expected" now?
Can we close the ticket?

berryma4, 2016-05-04:
Can we close this?

closed 2016-05-04 by kasemir