ControlSystemStudio/cs-studio #1520: Help documentation for diirt configuration

shroffk, 2015-12-15:


shroffk, 2015-12-15:
When cs-studio product is started the first diirt module to start up is the org.csstudio.diirt.util (This is the module that provides the description for the osgi services associated with datasources, functions, and formulas)

```
Dec 15, 2015 8:57:29 AM org.csstudio.diirt.util.Activator start
INFO: Starting diirt.util
```
Since the preferences are not yet available, diirt uses the config location defined in the jvm system properties, if you want a different default from the one described you can manually set the jvm property "diirt.home" in your products launch configuration

```
Dec 15, 2015 8:57:29 AM org.diirt.util.config.Configuration getFile
INFO: Loading C:\Users\Kunal Shroff\.diirt\datasources\datasources.xml
```

Following this all the Datasources Providers - provided via java services are made registered
```
Found bundle org.diirt.datasource_3.0.1.20151009125656 [880]
Found bundle org.eclipse.osgi.internal.framework.BundleContextImpl@43bf3668
Found bundles [Lorg.osgi.framework.Bundle;@4c6f307e
Found bundleentry://809.fwk741191984/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-loc_3.0.1.20151106105554 [809]
Providers [org.diirt.datasource.loc.LocalDataSourceProvider]
Provider instance org.diirt.datasource.loc.LocalDataSourceProvider@362ba6cc
Found bundleentry://884.fwk741191984/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sim_3.0.1.20151009125622 [884]
Providers [org.diirt.datasource.sim.SimulationDataSourceProvider]
Provider instance org.diirt.datasource.sim.SimulationDataSourceProvider@42427546
Found bundleentry://885.fwk741191984/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sys_3.0.1.20151009125624 [885]
Providers [org.diirt.datasource.sys.SystemDataSourceProvider]
Provider instance org.diirt.datasource.sys.SystemDataSourceProvider@69c2cc9c
Found bundleentry://889.fwk741191984/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.support.diirt-ca_3.0.1.20151009125648 [889]
Providers [org.diirt.support.ca.JCADataSourceProvider]
Provider instance org.diirt.support.ca.JCADataSourceProvider@51dae77a
```

Now the DatasourceProviders  aviable as osgi services are registered

```
!ENTRY org.eclipse.equinox.ds 1 0 2015-12-15 08:57:29.824
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterFormulaFunctionSet. The reference is: Reference[name = FormulaFunctionSet, interface = org.diirt.datasource.formula.FormulaFunctionSet, policy = dynamic, cardinality = 0..n, target = null, bind = registerFormulaFunctionSet, unbind = deregisterFormulaFunctionSet]

!ENTRY org.eclipse.equinox.ds 1 0 2015-12-15 08:57:29.844
!MESSAGE Could not bind a reference of component org.csstudio.diirt.util.RegisterService. The reference is: Reference[name = ServiceProvider, interface = org.diirt.service.ServiceProvider, policy = static, cardinality = 0..n, target = null, bind = registerService, unbind = null]

Dec 15, 2015 8:57:30 AM org.csstudio.diirt.util.RegisterDatasource registerDatasource
INFO: register:beast
```

Now that all the providers are registered, the o.c.diirt.util.preferences reads the cs-studio diirt preferences (Keep in mind that this can be done only after the workspace is available).

In the following case it is pointing to my test config folder with simulation (sim) set as the default datasource.

```
2015-12-15 08:57:35.970 CONFIG [Thread 1] org.csstudio.diirt.util.preferences.DiirtStartup (preWindowOpen) - Setting Diirt configuration folder to :C:\Users\Kunal Shroff\Desktop\test-config-folder
2015-12-15 08:57:37.430 CONFIG [Thread 1] org.csstudio.autocomplete.ui.AutoCompleteUIPlugin (readExtensionRegistry) - UI Helper provided by org.csstudio.autocomplete.ui.rcp
2015-12-15 08:57:37.701 INFO [Thread 1] org.diirt.util.config.Configuration (getFile) - Loading C:\Users\Kunal Shroff\.diirt\datasources\beast\beast.xml
2015-12-15 08:57:37.754 CONFIG [Thread 1] org.diirt.datasource.CompositeDataSource (retrieveDataSource) - Created instance for data source beast (BeastDataSource)
2015-12-15 08:57:38.109 INFO [Thread 50] org.apache.activemq.transport.failover.FailoverTransport (doReconnect) - Successfully connected to tcp://localhost:9102
2015-12-15 08:57:38.125 CONFIG [Thread 1] org.diirt.datasource.CompositeDataSource (retrieveDataSource) - Created instance for data source sim (SimulationDataSource)
2015-12-15 08:57:53.626 INFO [Thread 24] org.csstudio.alarm.beast.ui.clientmodel.AlarmClientModel (readConfiguration) - Read 1999 alarm tree items, 1807 PVs in 15.82 seconds: 126.3 items/sec, 114.2 PVs/sec
```

When the datasources are created they use the appropriate configuration folder as set by the cs-studio preferences.

shroffk, 2015-12-15:
@jbobnar @willrogers @berryma4  can you confirm if you see similar behavior,

jbobnar, 2015-12-15:
I confirm, except for the last sentence. When the datasources are created they use whatever is defined in diirt.home system variable. Because DiirtStartup hasn't been called yet, the preference was not yet read from the workspace and diirt.home has not been set yet. This means that diirt always defaults to the <user.home>/.diirt. It is impossible to configure CSS to use any other location, unless the -Ddiirt.home is specified in the launch configuration. This works, but it is not configurable via preferences dialog.

I don't always see this behaviour, but when I did, it stuck with me until I removed the entire workspace and launch configuration.

willrogers, 2015-12-15:
I have had the exact same experience as @jbobnar - Diirt is registered first so the preference selection does not take effect.

I've had to disable my datasource for this reason.

shroffk, 2015-12-15:
Are your data sources being created before the (PreWindowOpen) occurs?

shroffk, 2015-12-15:
@willrogers @jbobnar 
I have changed bits of the diirt.util such that the preference setting is now part of the plugin activation.
Currenlty I am not explicitly managing the bundle life cycle, could you test this change and see if it works with your datasource?

shroffk, 2015-12-15:
Now the order is as follows 

```
Dec 15, 2015 11:25:30 AM org.csstudio.diirt.util.Activator start
INFO: Starting diirt.util

Dec 15, 2015 11:25:30 AM org.csstudio.diirt.util.Activator setDiirtHome
INFO: Setting Diirt configuration folder to :C:\Users\Kunal Shroff\Desktop\test

Dec 15, 2015 11:25:30 AM org.diirt.util.config.Configuration getFile
INFO: Loading C:\Users\Kunal Shroff\Desktop\test\datasources\datasources.xml

Found bundle org.diirt.datasource_3.0.1.20151009125656 [880]
Found bundle org.eclipse.osgi.internal.framework.BundleContextImpl@4b5bcee9
Found bundles [Lorg.osgi.framework.Bundle;@6877c7bd
Found bundleentry://809.fwk1050149826/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-loc_3.0.1.20151106105554 [809]
Providers [org.diirt.datasource.loc.LocalDataSourceProvider]
Provider instance org.diirt.datasource.loc.LocalDataSourceProvider@6f6625a4
Found bundleentry://884.fwk1050149826/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sim_3.0.1.20151009125622 [884]
Providers [org.diirt.datasource.sim.SimulationDataSourceProvider]
Provider instance org.diirt.datasource.sim.SimulationDataSourceProvider@60679e19
Found bundleentry://885.fwk1050149826/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.datasource-sys_3.0.1.20151009125624 [885]
Providers [org.diirt.datasource.sys.SystemDataSourceProvider]
Provider instance org.diirt.datasource.sys.SystemDataSourceProvider@540c4b1a
Found bundleentry://889.fwk1050149826/META-INF/services/org.diirt.datasource.DataSourceProvider in org.diirt.support.diirt-ca_3.0.1.20151009125648 [889]
Providers [org.diirt.support.ca.JCADataSourceProvider]
Provider instance org.diirt.support.ca.JCADataSourceProvider@b6acd4c
Dec 15, 2015 11:25:30 AM org.csstudio.diirt.util.RegisterDatasource registerDatasource

INFO: register Datasource:beast
```

jbobnar, 2015-12-15:
@shroffk Where can I find this patch?

shroffk, 2015-12-15:
https://github.com/shroffk/cs-studio/commit/2f89a7fdaa7f6efbcda6b5c15beb0079c3eef9b4

jbobnar, 2015-12-15:
It works for me. Thanks!

shroffk, 2015-12-15:
Great, I will wait to see if this resolves the issues noticed by will

willrogers, 2015-12-16:
This seems to work for me too.

willrogers, 2015-12-16:
p.s. I built this using Maven and it works fine (I disabled the beast part because that's not building).

berryma4, 2015-12-18:
@shroffk 4.2.x branch?

shroffk, 2015-12-18:
Yes, this needs to be in 4.2

jbobnar, 2015-12-22:
Another problem. The fix causes the workspace prompt to stop working. org.csstudio.diirt.util instantiates the location, workspace prompt sees that workspace already exists and doesn't offer the dialog to select it.

shroffk, 2015-12-22:
Sorry, this was the reason why the initial version used the utility.product extension point...but then the java services and ds both happen before the workspace is created and thus fails to use preferences.

When I am back I shall look again into the startup process and see if I find a more appropriate point to initialize.

Maybe diirt needs to be rethought to allow reloading of datasources and/or providers.

kasemir, 2016-01-04:
That makes CSS unusable for SNS. We need the workspace selection on startup.
Can you revert the diirt change? Or is there chance to rework the diirt initialization in the next 2 weeks?

shroffk, 2016-01-04:
I agree this is a high priority issue, I am working on this now.

shroffk, 2016-01-05:
I have added a work around to this problem

In diirt I have changed the diirt configuration class to have a method that would re-read the configuration folder. 

In cs-studio, I moved the reading of the diirt preferences to the preWindowOpen startup stage, this would ensure that the workspace prompt starts working again.

This however has to be used carefully since the new configuration folder is only applied to the new instances of datasources that are created...in cs-studio the datasource providers are registered (both by osgi and java declaritive services) upon startup, but I update the configuration folder in the preWindowOpen step (before the datasources are instantiated).

berryma4, 2016-01-26:
@shroffk, can we close this  now?

closed 2016-01-26 by shroffk