ControlSystemStudio/cs-studio #1531: "Edit OPI" appears in all context menus

kasemir, 2015-12-17:
The "Edit OPI" menu item appears not only in the context menu of the OpiRuntime, where it makes sense, but also in the context menus of the Databrowser plot, PVTable, EPICS PV Tree, ...

The fix might be some incantation of `<visibleWhen>` checking if the `activePart` is  an org.csstudio.opibuilder.opiView, org.csstudio.opibuilder.opiViewTOP, ...

kasemir, 2015-12-17:
@willrogers  Changing the "Edit OPI" menu contribution to the following fixes the problem for the most part.

 * The "Edit OPI" menu entry does still show up in the OPIView
 * It's no longer in the Databrowser, PVTable, EPICS PV Tree, ..

.. but then it's also no longer in the context menu of the standalone OPI Shell..

```
   <extension point="org.eclipse.ui.menus">
      <menuContribution allPopups="false"
            locationURI="popup:org.eclipse.ui.popup.any?after=additions">
         <command  commandId="org.csstudio.opibuilder.commands.editopi"
               icon="icons/edit.gif"
               label="Edit OPI"
               style="push">
            <visibleWhen checkEnabled="true">
               <with variable="activePart">
                  <instanceof value="org.csstudio.opibuilder.runmode.OPIView"/>
               </with>
            </visibleWhen>
         </command>
      </menuContribution>
   </extension>
```

kasemir, 2015-12-17:
For what it's worth, the original behaviour is somewhat intermittent. 
Below are snapshots of the "Edit OPI" menu item appearing on non-OPI parts. Invoking the erroneous menu item once will result in message box "The chosen operation is not enabled" and it will then disappear until CSS is restarted.
But there are other times where it disappears. I have not determined what all makes it come and go.

[image]
[image]
[image]

willrogers, 2015-12-18:
Yes, we have noticed this.  We will take a look.

@nickbattam

kasemir, 2015-12-18:
Thanks for looking into it.
What might work:
1) Add the `menuContribution ..command  ..visibleWhen ..activePart .. instanceof ..OPIView` so that it works OK for all the RCP code (shows in OPIView, does not show erroneously elsewhere).
2) Add another solution for the OPI Shell. Since it's less integrated with RCP, none of the https://wiki.eclipse.org/Command_Core_Expressions may apply, but one could still try to hinge upon activeShell. If all fails, the OPIShell might have to add its own expression variable "opiShellActive" via http://www.vogella.com/tutorials/EclipseCommandsAdvanced/article.html#commmands_sourceprovider

willrogers, 2016-09-23:
We think this is fixed.  Please close the ticket unless you know otherwise.

closed 2016-09-28 by berryma4