ControlSystemStudio/cs-studio #1533: BOY examples need version upped?

berryma4, 2015-12-18:
The BOY examples give this error:

[image]

Is this from the version up we did?  Do they need to be in sync?

kasemir, 2015-12-18:
I have always seen that error because the example main.opi does an exact `if(opiVersion.compareTo(BOYVersion)<0)`, so each nightly build with its time stamp qualifier results in 'outdated' examples.

Something like `if(opiVersion.getMajor() < BOYVersion.getMajor())` would be more tolerant, but I'm also OK with simply removing that script from the main.opi, because the examples are in fact outdated, #767, and that's not just a matter of version numbers.

berryma4, 2015-12-21:
The plugin version is 5.0.0-snapshot for o.c.opibuilder.examples.  Where does this 4.0.999 come from?

berryma4, 2015-12-21:
Ah, I see, it's in the BOY example file itself.

berryma4, 2015-12-21:
I added the ```if(opiVersion.getMajor() < BOYVersion.getMajor())```

closed 2015-12-21 by berryma4