ControlSystemStudio/cs-studio #1544: GTK 3

berryma4, 2016-01-04:
Does anyone have gtk3 working with Eclipse Mars (CS-Studio 4.2+)?
I'm on Debian Jessie.

I get an abort very soon after the workbench comes up at gdk_display_open:
``` gdb
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `/usr/bin/java -Xmx1024m -Xms128m -Dorg.eclipse.update.reconcile=false -Dcom.cos'.
Program terminated with signal SIGABRT, Aborted.
#0  0x00007ff729f29107 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	../nptl/sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) where
#0  0x00007ff729f29107 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ff729f2a4e8 in __GI_abort () at abort.c:89
#2  0x00007ff7298356b5 in os::abort(bool) () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/server/libjvm.so
#3  0x00007ff7299d3bf3 in VMError::report_and_die() () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/server/libjvm.so
#4  0x00007ff72983aedf in JVM_handle_linux_signal () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/server/libjvm.so
#5  0x00007ff729831673 in signalHandler(int, siginfo*, void*) () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/server/libjvm.so
#6  <signal handler called>
#7  0x00007ff6c625f262 in gdk_display_open () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#8  0x00007ff6c622f01d in gdk_display_open_default_libgtk_only () from /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0
#9  0x00007ff6c65f7422 in gtk_init_check () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#10 0x00007ff6c65f7439 in gtk_init () from /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0
#11 0x00007ff6bfb3758e in JNI_OnLoad () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/libglass.so
#12 0x00007ff7285e4d07 in Java_java_lang_ClassLoader_00024NativeLibrary_load () from /opt/jdk/jdk1.8.0_65/jre/lib/amd64/libjava.so
#13 0x00007ff7149e3994 in ?? ()
#14 0x00007ff72aacfa50 in ?? ()
#15 0x00000000c07cd2c8 in ?? ()
#16 0x00000000eb0c81e0 in ?? ()
#17 0x0000000000000019 in ?? ()
#18 0x00007ff72aacfa10 in ?? ()
#19 0x0000000000000000 in ?? ()

```

berryma4, 2016-01-04:
interesting enough, the general / common product works fine with gtk3.  I will look at my products features in more detail.

kasemir, 2016-01-04:
We do use it, but with with SWT options to disable Cairo.

Under Help/About/... it looks like this:

org.eclipse.swt.internal.gtk.cairoGraphics=false
org.eclipse.swt.internal.gtk.useCairo=false
org.eclipse.swt.internal.gtk.version=3.14.13

berryma4, 2016-01-04:
The culprit seems to be a pop up from jGit, asking for a username password.

berryma4, 2016-01-04:
@shroffk Log Table looks a little strange under GTK3
[image]

kasemir, 2016-01-04:
In our case, *.opi displays looked different because opibuilder treats font sizes as points, so pixel size changes with DPIs of the screen. Had to adjust the fonts.def to work around it.

berryma4, 2016-01-04:
@kasemir, do you use the "Logging Configuration" UI?
This seems to give me the same gdk_display_open crash the jgit pop up did.

berryma4, 2016-01-04:
@kasemir would you be willing to share your fonts.def as a guide?

berryma4, 2016-01-04:
I was thinking of pushing Liberation fonts here, because they are on Debian Jessie ... and I have windows users going to linux.
https://en.wikipedia.org/wiki/Liberation_fonts

berryma4, 2016-01-04:
okay, I wasn't seeing the issues from efxclipse because you need to add to cs-studio.ini:
```
-Defxclipse.osgi.hook.debug=true
```

so now I get for the jgit pop up:
```
log4j:WARN No appenders could be found for logger (org.eclipse.jgit.util.FS).
log4j:WARN Please initialize the log4j system properly.
WARNING: You are binding against the deprecated org.eclipse.fx.javafx - please remove all javafx imports
FXClassLoader#createJREBundledClassloader - Started
FXClassLoader#createJREBundledClassloader - Assumed location (Java 8/Java 7): /opt/jdk/jdk1.8.0_65/jre/lib/ext/jfxrt.jar
FXClassLoader#createJREBundledClassloader - SWT is available use different loading strategy
FXClassLoader#createJREBundledClassloader - Searching for SWT-FX integration at /opt/jdk/jdk1.8.0_65/jre/lib/jfxswt.jar
FXClassLoader#createJREBundledClassloader - Found SWT/FX
FXClassLoader#createJREBundledClassloader - Delegate to system classloader
FXClassLoader.SWTFXClassloader#init - Primary Loader sun.misc.Launcher$ExtClassLoader@42220171
FXClassLoader.SWTFXClassloader#init - Lastresort Loader org.eclipse.osgi.internal.loader.EquinoxClassLoader@36ac8a63[org.eclipse.swt:3.104.1.v20150825-0743(id=344)]
FXClassLoader#createJREBundledClassloader - Ended
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.Node'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.control.TextField'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.control.PasswordField'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.control.Label'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.control.CheckBox'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.Parent'
FXModuleClassloader#findLocalClass trying to load class 'javafx.scene.layout.GridPane'
FXModuleClassloader#findLocalClass - Someone is trying to load FXCanvas. Need to check for GTK3
FXModuleClassloader#findLocalClass - We are on GTK need to take a closer look
FXModuleClassloader#findLocalClass - We are on GTK3 - too bad need to disable JavaFX for now else we'll crash the JVM
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Canvas with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Canvas with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.widgets.Canvas class org.eclipse.swt.widgets.Canvas
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.graphics.Device with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.graphics.Device with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.graphics.Device class org.eclipse.swt.graphics.Device
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Display with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Display with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.widgets.Display class org.eclipse.swt.widgets.Display
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.Transfer with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.Transfer with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.dnd.Transfer class org.eclipse.swt.dnd.Transfer
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.ByteArrayTransfer with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.ByteArrayTransfer with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.dnd.ByteArrayTransfer class org.eclipse.swt.dnd.ByteArrayTransfer
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.DisposeListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.DisposeListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.DisposeListener interface org.eclipse.swt.events.DisposeListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.MouseListener interface org.eclipse.swt.events.MouseListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseTrackListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseTrackListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.MouseTrackListener interface org.eclipse.swt.events.MouseTrackListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.ControlListener with primary
[... cut ...]
# C  [libgdk-x11-2.0.so.0+0x4e262]  gdk_display_open+0x52
#
# Core dump written. Default location: /home/physuser/Desktop/cs-studio/core or core.10630
#
# An error report file with more information is saved as:
# /home/physuser/Desktop/cs-studio/hs_err_pid10630.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

berryma4, 2016-01-04:
and similar for the Logging Configuration UI:
```
FXClassLoader#createJREBundledClassloader - Started
FXClassLoader#createJREBundledClassloader - Assumed location (Java 8/Java 7): /opt/jdk/jdk1.8.0_65/jre/lib/ext/jfxrt.jar
FXClassLoader#createJREBundledClassloader - SWT is available use different loading strategy
FXClassLoader#createJREBundledClassloader - Searching for SWT-FX integration at /opt/jdk/jdk1.8.0_65/jre/lib/jfxswt.jar
FXClassLoader#createJREBundledClassloader - Found SWT/FX
FXClassLoader#createJREBundledClassloader - Delegate to system classloader
FXClassLoader.SWTFXClassloader#init - Primary Loader sun.misc.Launcher$ExtClassLoader@12454f13
FXClassLoader.SWTFXClassloader#init - Lastresort Loader org.eclipse.osgi.internal.loader.EquinoxClassLoader@32f61a31[org.eclipse.swt:3.104.1.v20150825-0743(id=344)]
FXClassLoader#createJREBundledClassloader - Ended
FXModuleClassloader#findLocalClass - Someone is trying to load FXCanvas. Need to check for GTK3
FXModuleClassloader#findLocalClass - We are on GTK need to take a closer look
FXModuleClassloader#findLocalClass - We are on GTK3 - too bad need to disable JavaFX for now else we'll crash the JVM
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Canvas with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Canvas with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.widgets.Canvas class org.eclipse.swt.widgets.Canvas
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.graphics.Device with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.graphics.Device with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.graphics.Device class org.eclipse.swt.graphics.Device
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Display with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.widgets.Display with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.widgets.Display class org.eclipse.swt.widgets.Display
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.Transfer with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.Transfer with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.dnd.Transfer class org.eclipse.swt.dnd.Transfer
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.ByteArrayTransfer with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.dnd.ByteArrayTransfer with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.dnd.ByteArrayTransfer class org.eclipse.swt.dnd.ByteArrayTransfer
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.DisposeListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.DisposeListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.DisposeListener interface org.eclipse.swt.events.DisposeListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.MouseListener interface org.eclipse.swt.events.MouseListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseTrackListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.MouseTrackListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.MouseTrackListener interface org.eclipse.swt.events.MouseTrackListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.ControlListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.ControlListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.ControlListener interface org.eclipse.swt.events.ControlListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.FocusListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.FocusListener with last resort
FXClassLoader.SWTFXClassloader#findClass - Result org.eclipse.swt.events.FocusListener interface org.eclipse.swt.events.FocusListener
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.KeyListener with primary
FXClassLoader.SWTFXClassloader#findClass - ClassNotFoundException thrown
FXClassLoader.SWTFXClassloader#findClass - Loading org.eclipse.swt.events.KeyListener with last resort
[... cut ...]
# C  [libgdk-x11-2.0.so.0+0x4e262]  gdk_display_open+0x52
#
# Core dump written. Default location: /home/physuser/Desktop/cs-studio/core or core.10711
#
# An error report file with more information is saved as:
# /home/physuser/Desktop/cs-studio/hs_err_pid10711.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

berryma4, 2016-01-04:
Well, 
https://bugs.openjdk.java.net/browse/JDK-8087516

it looks like if we want javafx, we have to run under gtk2

berryma4, 2016-01-04:
Because of reading so many conflicting threads, I though I would add a trace to gtk2 from jfxswt.jar

```
(physenv)physuser@develop-vmasd0:~/Desktop/cs-studio$ /opt/jdk/jdk1.8.0_65/bin/jdeps /opt/jdk/jdk1.8.0_65/jre/lib/jfxswt.jar
jfxswt.jar -> /opt/jdk/jdk1.8.0_65/jre/lib/ext/jfxrt.jar
jfxswt.jar -> not found
jfxswt.jar -> /opt/jdk/jdk1.8.0_65/jre/lib/rt.jar
   javafx.embed.swt (jfxswt.jar)
      -> com.sun.glass.ui                                   JDK internal API (jfxrt.jar)
      -> com.sun.javafx.application                         JDK internal API (jfxrt.jar)
      -> com.sun.javafx.cursor                              JDK internal API (jfxrt.jar)
      -> com.sun.javafx.embed                               JDK internal API (jfxrt.jar)
      -> com.sun.javafx.stage                               JDK internal API (jfxrt.jar)
      -> java.io                                            
      -> java.lang                                          
      -> java.lang.invoke                                   
      -> java.lang.reflect                                  
      -> java.nio                                           
      -> java.security                                      
      -> java.util                                          
      -> javafx.application                                 
      -> javafx.beans                                       
      -> javafx.scene                                       
      -> javafx.scene.image                                 
      -> javafx.scene.input                                 
      -> javafx.util                                        
      -> org.eclipse.swt                                    not found
      -> org.eclipse.swt.dnd                                not found
      -> org.eclipse.swt.events                             not found
      -> org.eclipse.swt.graphics                           not found
      -> org.eclipse.swt.widgets                            not found
(physenv)physuser@develop-vmasd0:~/Desktop/cs-studio$ ldd /opt/jdk/jdk1.8.0_65/jre/lib/amd64/libglass.so 
	linux-vdso.so.1 (0x00007fffc3edb000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f2421334000)
	libgdk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk-x11-2.0.so.0 (0x00007f242107e000)
	libatk-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libatk-1.0.so.0 (0x00007f2420e57000)
	libgio-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgio-2.0.so.0 (0x00007f2420ade000)
	libpangoft2-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangoft2-1.0.so.0 (0x00007f24208c8000)
	libgdk_pixbuf-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgdk_pixbuf-2.0.so.0 (0x00007f24206a6000)
	libpangocairo-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpangocairo-1.0.so.0 (0x00007f2420499000)
	libcairo.so.2 => /usr/lib/x86_64-linux-gnu/libcairo.so.2 (0x00007f2420184000)
	libpango-1.0.so.0 => /usr/lib/x86_64-linux-gnu/libpango-1.0.so.0 (0x00007f241ff36000)
	libfreetype.so.6 => /usr/lib/x86_64-linux-gnu/libfreetype.so.6 (0x00007f241fc8c000)
	libfontconfig.so.1 => /usr/lib/x86_64-linux-gnu/libfontconfig.so.1 (0x00007f241fa4f000)
	libgobject-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0 (0x00007f241f7fd000)
	libgmodule-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0 (0x00007f241f5f9000)
	libgthread-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f241f3f7000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f241f1ef000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f241eee0000)
	libXtst.so.6 => /usr/lib/x86_64-linux-gnu/libXtst.so.6 (0x00007f241ecda000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f241e9cf000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f241e6ce000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f241e4b8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f241e29b000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f241def2000)
	libX11.so.6 => /usr/lib/x86_64-linux-gnu/libX11.so.6 (0x00007f241dbaf000)
	libXcomposite.so.1 => /usr/lib/x86_64-linux-gnu/libXcomposite.so.1 (0x00007f241d9ac000)
	libXdamage.so.1 => /usr/lib/x86_64-linux-gnu/libXdamage.so.1 (0x00007f241d7a9000)
	libXfixes.so.3 => /usr/lib/x86_64-linux-gnu/libXfixes.so.3 (0x00007f241d5a3000)
	libXrender.so.1 => /usr/lib/x86_64-linux-gnu/libXrender.so.1 (0x00007f241d399000)
	libXinerama.so.1 => /usr/lib/x86_64-linux-gnu/libXinerama.so.1 (0x00007f241d196000)
	libXi.so.6 => /usr/lib/x86_64-linux-gnu/libXi.so.6 (0x00007f241cf86000)
	libXrandr.so.2 => /usr/lib/x86_64-linux-gnu/libXrandr.so.2 (0x00007f241cd7c000)
	libXcursor.so.1 => /usr/lib/x86_64-linux-gnu/libXcursor.so.1 (0x00007f241cb71000)
	libXext.so.6 => /usr/lib/x86_64-linux-gnu/libXext.so.6 (0x00007f241c95f000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f241c744000)
	libselinux.so.1 => /lib/x86_64-linux-gnu/libselinux.so.1 (0x00007f241c51f000)
	libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007f241c308000)
	libharfbuzz.so.0 => /usr/lib/x86_64-linux-gnu/libharfbuzz.so.0 (0x00007f241c0b1000)
	libpng12.so.0 => /lib/x86_64-linux-gnu/libpng12.so.0 (0x00007f241be8a000)
	libpixman-1.so.0 => /usr/lib/x86_64-linux-gnu/libpixman-1.so.0 (0x00007f241bbdd000)
	libxcb-shm.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-shm.so.0 (0x00007f241b9d9000)
	libxcb-render.so.0 => /usr/lib/x86_64-linux-gnu/libxcb-render.so.0 (0x00007f241b7cf000)
	libxcb.so.1 => /usr/lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f241b5ad000)
[... cut ...]
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f241ad6f000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f241ab01000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f2421bad000)
	libgraphite2.so.3 => /usr/lib/x86_64-linux-gnu/libgraphite2.so.3 (0x00007f241a8e5000)
	libXau.so.6 => /usr/lib/x86_64-linux-gnu/libXau.so.6 (0x00007f241a6e1000)
	libXdmcp.so.6 => /usr/lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f241a4dc000)
	libdatrie.so.1 => /usr/lib/x86_64-linux-gnu/libdatrie.so.1 (0x00007f241a2d4000)
```

berryma4, 2016-01-04:
So, the work around stays in place to force gtk 2:
add:
--launcher.GTK_version
2

to your product ini.

Or, 
#Gtk3 forced (crashes with javafx):
export SWT_GTK3=1
 
#Gtk2 forced:
export SWT_GTK3=0

closing ticket.

closed 2016-01-04 by berryma4

berryma4, 2016-01-04:
.... kind of funny, someone was assigned the AWT part on Christmas day:
https://bugs.openjdk.java.net/browse/JDK-8145547
Merry Christmas!!!!

kasemir, 2016-01-04:
Regarding the fonts.def, Liberation fonts as you already suggested are a good idea.
For the display builder update the predefined fonts are 
```
Default      = Liberation Sans - regular - 14
Default Bold = Liberation Sans - bold    - 14
Header 1     = Liberation Sans - bold    - 22
Header 2     = Liberation Sans - bold    - 18
Header 3     = Liberation Sans - bold    - 16
Comment      = Liberation Sans - italic  - 14
Fine Print   = Liberation Sans - regular - 12
```

For the  existing opibuilder, sharing that doesn't make sense because the font support is broken. With points instead of pixels, the handling of fonts can depend on the monitor that you plug into the computer.