ControlSystemStudio/cs-studio #1545: EDM converter creates spurious alarm sensitivity

willrogers, 2016-01-05:
BOY widgets honour alarm sensitivity even if no PV is set, so that such a widget shows as disconnected.  I think this behaviour is ok.

EDM doesn't do this.

I've changed the converter because EDM screens that accidentally do this now have lots of disconnected widgets when they were never supposed to have any PVs.  The change doesn't propagate the alarm sensitivity if there is no relevant PV.

kasemir, 2016-01-05:
What types of widgets do you have in EDM that have no PVs, but can have PVs in BOY?

willrogers, 2016-01-05:
That's not what I mean.  I have a screen full of EDM 'Static Text' widgets, which happen to have 'Bg Alarm Sensitive' checked, but have no PV in either 'Color PV' or 'Visibility PV'.  

EDM ignores this; the CSS converter creates a rule listening to the PV '' and reports the widget as disconnected.

kasemir, 2016-01-05:
Understand. In BOY, even "static" widgets can have a PV Name.
If none is provided, yet the widget is still configured to be alarm sensitive, I agree with you that BOY is probably correctly showing the disconnected state, so how you handle that in the EDM translator sounds good.

closed 2023-10-11 by willrogers