ControlSystemStudio/cs-studio #1557: Text Update vs Label fonts

berryma4, 2016-01-08:
[image]

In the above screenshot, both black Labels and blue Text Updates fonts are supposedly Arial 9 (Regular).  But, the sizes are very different.  What should I know about the difference of these two widgets?

This is on Debian Jessie, Xfce, 4K monitor, and Arial mapped to "Noto Sans":
```
$ fc-match 'Arial'
NotoSans-Regular.ttf: "Noto Sans" "Regular"
```

berryma4, 2016-01-08:
``` xml
<widget typeId="org.csstudio.opibuilder.widgets.Label" version="1.0.0">
      <visible>true</visible>
      <vertical_alignment>1</vertical_alignment>
      <wuid>5a70e74c:13fa4e4479b:-7a31</wuid>
      <auto_size>false</auto_size>
      <scripts />
      <height>20</height>
      <name>Label</name>
      <scale_options>
        <width_scalable>true</width_scalable>
        <height_scalable>true</height_scalable>
        <keep_wh_ratio>false</keep_wh_ratio>
      </scale_options>
      <transparent>true</transparent>
      <show_scrollbar>false</show_scrollbar>
      <background_color>
        <color red="255" green="255" blue="255" />
      </background_color>
      <foreground_color>
        <color red="0" green="0" blue="0" />
      </foreground_color>
      <widget_type>Label</widget_type>
      <enabled>true</enabled>
      <text>Radius</text>
      <font>
        <opifont.name fontName="Arial" height="9" style="0">Labels and Graph Axis</opifont.name>
      </font>
      <width>43</width>
      <border_style>0</border_style>
      <rules />
      <border_width>1</border_width>
      <border_color>
        <color red="0" green="128" blue="255" />
      </border_color>
      <horizontal_alignment>0</horizontal_alignment>
      <actions hook="false" hook_all="false" />
      <y>18</y>
      <wrap_words>true</wrap_words>
      <tooltip></tooltip>
      <x>12</x>
    </widget>
```

```xml
<widget typeId="org.csstudio.opibuilder.widgets.TextUpdate" version="1.0.0">
      <border_alarm_sensitive>true</border_alarm_sensitive>
      <visible>true</visible>
      <vertical_alignment>1</vertical_alignment>
      <show_units>true</show_units>
      <wuid>5a70e74c:13fa4e4479b:-7a22</wuid>
      <auto_size>false</auto_size>
      <rotation_angle>0.0</rotation_angle>
      <scripts />
      <height>20</height>
      <name>Text Update_6</name>
      <forecolor_alarm_sensitive>false</forecolor_alarm_sensitive>
      <scale_options>
        <width_scalable>true</width_scalable>
        <height_scalable>true</height_scalable>
        <keep_wh_ratio>false</keep_wh_ratio>
      </scale_options>
      <format_type>0</format_type>
      <precision_from_pv>true</precision_from_pv>
      <transparent>false</transparent>
      <pv_name>$(PREFIX):MOTORRAD.RBV</pv_name>
      <background_color>
        <color red="255" green="255" blue="255" />
      </background_color>
      <foreground_color>
        <color name="ReadBack/Graphs Foreground" red="0" green="80" blue="220" />
      </foreground_color>
      <widget_type>Text Update</widget_type>
      <enabled>true</enabled>
      <text>######</text>
      <backcolor_alarm_sensitive>false</backcolor_alarm_sensitive>
      <precision>0</precision>
      <font>
        <opifont.name fontName="Arial" height="9" style="0">Read backs and Input Fields</opifont.name>
      </font>
      <width>73</width>
      <border_style>0</border_style>
      <rules />
      <pv_value />
      <border_width>1</border_width>
      <border_color>
        <color red="0" green="128" blue="255" />
      </border_color>
      <horizontal_alignment>0</horizontal_alignment>
      <actions hook="false" hook_all="false" />
      <y>18</y>
      <wrap_words>false</wrap_words>
      <tooltip>$(pv_name)
$(pv_value)</tooltip>
      <x>54</x>
    </widget>
```

kasemir, 2016-01-08:
Are these in a group which scales its content?
Maybe try to reproduce in a display that has only one label and one textupdate.

berryma4, 2016-01-11:
Like I said, this is a 4k monitor.  I would understand font issues, but I didn't expect this.
[image]

berryma4, 2016-01-11:
file:
```xml
<?xml version="1.0" encoding="UTF-8"?>
<display typeId="org.csstudio.opibuilder.Display" version="1.0.0">
  <actions hook="false" hook_all="false" />
  <auto_scale_widgets>
    <auto_scale_widgets>false</auto_scale_widgets>
    <min_width>-1</min_width>
    <min_height>-1</min_height>
  </auto_scale_widgets>
  <auto_zoom_to_fit_all>false</auto_zoom_to_fit_all>
  <background_color>
    <color name="Screen Background" red="240" green="240" blue="240" />
  </background_color>
  <boy_version>5.0.0.201512241344</boy_version>
  <foreground_color>
    <color red="192" green="192" blue="192" />
  </foreground_color>
  <grid_space>6</grid_space>
  <height>600</height>
  <macros>
    <include_parent_macros>true</include_parent_macros>
  </macros>
  <name></name>
  <rules />
  <scripts />
  <show_close_button>true</show_close_button>
  <show_edit_range>true</show_edit_range>
  <show_grid>true</show_grid>
  <show_ruler>true</show_ruler>
  <snap_to_geometry>true</snap_to_geometry>
  <widget_type>Display</widget_type>
  <width>800</width>
  <wuid>52f5b9fd:14150e27666:-56aa</wuid>
  <x>-1</x>
  <y>-1</y>
  <widget typeId="org.csstudio.opibuilder.widgets.TextUpdate" version="1.0.0">
    <actions hook="false" hook_all="false" />
    <alarm_pulsing>false</alarm_pulsing>
    <auto_size>false</auto_size>
    <backcolor_alarm_sensitive>false</backcolor_alarm_sensitive>
    <background_color>
      <color red="255" green="255" blue="255" />
    </background_color>
    <border_alarm_sensitive>true</border_alarm_sensitive>
    <border_color>
      <color red="0" green="128" blue="255" />
    </border_color>
    <border_style>0</border_style>
    <border_width>1</border_width>
    <enabled>true</enabled>
    <font>
      <opifont.name fontName="Arial" height="9" style="0">Read backs and Input Fields</opifont.name>
    </font>
    <forecolor_alarm_sensitive>false</forecolor_alarm_sensitive>
    <foreground_color>
      <color red="0" green="0" blue="0" />
    </foreground_color>
    <format_type>0</format_type>
    <height>20</height>
    <horizontal_alignment>0</horizontal_alignment>
    <name>Text Update</name>
    <precision>0</precision>
    <precision_from_pv>true</precision_from_pv>
    <pv_name></pv_name>
    <pv_value />
    <rotation_angle>0.0</rotation_angle>
    <rules />
    <scale_options>
      <width_scalable>true</width_scalable>
      <height_scalable>true</height_scalable>
      <keep_wh_ratio>false</keep_wh_ratio>
    </scale_options>
    <scripts />
    <show_units>true</show_units>
    <text>Text</text>
    <tooltip>$(pv_name)
$(pv_value)</tooltip>
    <transparent>false</transparent>
    <vertical_alignment>1</vertical_alignment>
    <visible>true</visible>
    <widget_type>Text Update</widget_type>
    <width>100</width>
    <wrap_words>false</wrap_words>
    <wuid>-23bdadde:1523147a3cc:-7fa0</wuid>
    <x>330</x>
    <y>60</y>
  </widget>
  <widget typeId="org.csstudio.opibuilder.widgets.Label" version="1.0.0">
    <actions hook="false" hook_all="false" />
    <auto_size>false</auto_size>
    <background_color>
      <color red="255" green="255" blue="255" />
    </background_color>
    <border_color>
      <color red="0" green="128" blue="255" />
    </border_color>
    <border_style>0</border_style>
    <border_width>1</border_width>
    <enabled>true</enabled>
    <font>
      <opifont.name fontName="Arial" height="9" style="0">Read backs and Input Fields</opifont.name>
    </font>
    <foreground_color>
      <color red="0" green="0" blue="0" />
    </foreground_color>
    <height>20</height>
    <horizontal_alignment>1</horizontal_alignment>
    <name>Label</name>
    <rules />
    <scale_options>
      <width_scalable>true</width_scalable>
      <height_scalable>true</height_scalable>
      <keep_wh_ratio>false</keep_wh_ratio>
    </scale_options>
    <scripts />
    <text>Text</text>
    <tooltip></tooltip>
    <transparent>true</transparent>
    <vertical_alignment>1</vertical_alignment>
    <visible>true</visible>
    <widget_type>Label</widget_type>
    <width>80</width>
    <wrap_words>false</wrap_words>
    <wuid>-23bdadde:1523147a3cc:-7f9c</wuid>
    <x>222</x>
    <y>60</y>
  </widget>
</display>
```

kasemir, 2016-01-11:
Can't reproduce that here, which makes it even more bizarre.
Looking at the code, both LabelEditPart and TextUpdateEditPart should create the same TextFigure for wrap_words==false. In TextFigure#paintFigure() is this code, which may trigger in your case:
```
//shrink font size to fit the figure.
int i=0        
while(h > (clientArea.height+2) && h > 10 && i++<20){
     FontData fd = font.getFontData()[0];
     fd.setHeight(fd.getHeight()-1);
     font = CustomMediaFactory.getInstance().getFont(fd);
     h = getTextSize(font).height;
}
```

You might have to put a breakpoint or printout into TextFigure#paintFigure() to see if that's the case.
Otherwise, where it sets the font, see if the font ends up being different.

berryma4, 2016-01-11:
I do get an exception for this for the user screen, but not the simplified (attached) one text box, one input.

``` java
Caused by: java.lang.IllegalArgumentException: Argument not valid
	at org.eclipse.swt.SWT.error(SWT.java:4472)
	at org.eclipse.swt.SWT.error(SWT.java:4406)
	at org.eclipse.swt.SWT.error(SWT.java:4377)
	at org.eclipse.swt.graphics.FontData.setHeight(FontData.java:340)
	at org.csstudio.swt.widgets.figures.TextFigure.paintFigure(TextFigure.java:329)
	at org.eclipse.draw2d.Figure.paint(Figure.java:1118)
```

kasemir, 2016-01-11:
That's a hint. At least in the user screen, you are invoking the "shrink font size to fit the figure" code, since TextFigure.java:329 is the fd.setHeight() in there.

So the overall scenario could be:
On the different display with 4K monitor, the (unfortunately) "point" based font sizes come out larger than before. Based on the widget size, the "shrink font size to fit the figure" code is invoked and the font gets shrunken.

The real fix would be to update the opibuilder font handling to be pixel-based. Since the font API ist still point based, it could for example use code as in the display.builder to calibrate the pixels/point mapping.
For the display.builder I chose that route because it's clearly a new tool.
For the opibuilder I'm not sure, because it _will_ change the font sizes of most existing displays. They're consistent from then on, but as you change to pixel based fonts you do have a regression.

berryma4, 2016-01-11:
Simply commenting out TextFigure.java:329 shows this is the issue (other than the input also droops a bit lower now in Linux).  I will connect the 4k monitor to my windows box, just to confirm it has nothing to do with linux/gtk.

Linux:
[image]

Windows:
[image]

berryma4, 2016-01-11:
After adjusting the fonts.conf file (reduce size by 60%), ... they look the same now.  I will probably have to adjust the gtkrc file to make up for BOY with the menu, etc.

Linux:
[image]

it seems like these lines should be an option, not the default, no?:
```java
//shrink font size to fit the figure.
int i=0        
while(h > (clientArea.height+2) && h > 10 && i++<20){
     FontData fd = font.getFontData()[0];
     fd.setHeight(fd.getHeight()-1);
     font = CustomMediaFactory.getInstance().getFont(fd);
     h = getTextSize(font).height;
}
```

I'm not quite sure what to do, that's a bad regression problem.

kasemir, 2016-01-11:
The BOY font handling is broken, and any fix will be a regression.

For the display.builder I treat the font sizes as pixels, so going forward I get the same sizes on the screen.
But then it's still unclear what to do when reading legacy files. For the SNS displays, that scaling factor is about 0.75 based on the Linux setups we have at beam lines. That would have to be a configuration setting for each control room.

Meanwhile, if you use named fonts (font 'macros'), you can adjust the font sizes via different font.def files.
Simply shrinking the text in a trial-and-error fashion is certainly odd. Maybe your suggestion to remove that code is most practical for BOY, unless somebody has time to update the complete font handling. I'm happy to explain the code in the display.builder, but I'd rather go forward there and not backport that into BOY.

berryma4, 2016-01-11:
Maybe a system property to skip the code?
This would be similar to the wonderful IE Compatibility View.

kasemir, 2016-01-11:
If you want to make it selectable, I'm OK with by default skipping the code.

berryma4, 2016-01-11:
I am pretty happy, now that I adjusted the gtkrc file for the menus (I made them a little larger than the windows version).

Linux:
[image]

Windows:
[image]

It's a good point that I could just change the BOY fonts.def.  I'm just afraid a font somewhere will default to something huge (but, I guess this could always happen).

berryma4, 2016-02-11:
I fixed this by having special fonts.conf and gtkrc files loaded inline with the cs-studio command.
(so, it doesn't change the whole linux desktop)

berryma4, 2016-02-11:
This is a larger issue, as Kay mentioned, ... I'm not in favor of little tweaks here and there in BOY, when fonts in general need to be reworked.

closed 2016-02-11 by berryma4