ControlSystemStudio/cs-studio #1568: OPI Editor perspective should open if you open an OPI file

willrogers, 2016-01-15:
First-time users of CS-Studio typically go to File | New | OPI File and create an OPI.

An OPI editor opens, but the properties view isn't there.  It kind of looks like you can create an OPI, but you really can't.

This applies to me too: I often open an OPI file in the editor but then need to manually select the OPI Editor perspective.

I think OPI editor perspective should be chosen automatically if you open an OPI file in the editor.

kasemir, 2016-01-15:
I also often need to select the editor perspective manually as soon as I edit a display file.
But previous attempts to automatically switch perspectives for other cases (data browser, opi view) turned out to cause more harm than good.
The only case where I encounter the IDE switching perspectives on me is for debugging, where it's a "Do you want to switch.." dialog.

So maybe do that: Prompt for a switch, which still allows a user to decline if that's not what you want.

willrogers, 2016-01-15:
I think you're right, it's got to be one of those options (maybe 'Always do this?' or 'Never do this') attached to a preference somewhere.

Debugging in Eclipse is the classic example.

willrogers, 2016-01-26:
I can do this easily in the `init()` method of the editor, but:

* this isn't really the right place
* it prompts also on startup, which is unfriendly

Do you know of a better way to intercept the opening of a particular editor?

kasemir, 2016-01-26:
Well, trying to guess which perspective the user needs is difficult..

Still, this is for the editor, so people who simply execute existing displays won't be affected, only those who really need to edit displays. I can't think of a place other than the editor's init(). Maybe do this:
```
 .. init(..)
{
  IPerspectiveDescriptor persp =
        PlatformUI.getWorkbench().getActiveWorkbenchWindow()
                         .getActivePage().getPerspective();
  if (! persp.getId().equals("..opi editor perspective ID"))
  {
      .. dialog "Do you want to switch to the OPI editor perspective" ..
```

That way, those who do want to edit will get the prompt.
On startup, if you had the editor open on shutdown, that was very likely in the OPI editor perspective, so you won't be bothered with another dialog.

willrogers, 2016-01-26:
I asked the Dawn people for ideas, and there is a way to do it without having it in the `init()` method.  I think it's less invasive this way, and the preference could be set by a site to default to "Never", meaning that no-one sees the behaviour by default.

willrogers, 2016-09-23:
This is now done - feel free to close.

closed 2016-09-23 by willrogers