ControlSystemStudio/cs-studio #1569: Maven build for version 4.2 and master failed

CTschentscher, 2016-01-18:
Hello everybody,
I'm not sure to be right here with my question.

I used Java Version 1.8.0.66 from Oracle and Maven version 3.3.9.
The build of maven-osgi-bundles, cs-studio-tgirdparty and dirt are success.
Cs-studio.core failed wirth:
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier requires 'package org.diirt.vtype 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT........
Csstudio.applications also failed:
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.alarm.beast 4.1.1.qualifier
[ERROR]   Missing requirement: org.csstudio.alarm.beast 4.1.1.qualifier requires 'bundle org.csstudio.csdata 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.alarm.beast:4.1.1-SNAPSHOT.....

As a consequence of this also org.csstudio.product failed.

I'm grateful for each hint.

kasemir, 2016-01-19:
Hi:

How to build CSS from sources, now with Maven/Tycho, keeping up with package movements and recently the rearrangement of the pvmanager (first removed from CSS, then renamed to diirt, now back in https://github.com/ControlSystemStudio but as its own repo, .. ) can be difficult.

A comparably short example of how to compile products can be found in
https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build

It’s what I use for the nightly build of the SNS products.

From your error message I assume that you’re somehow missing the diirt plugins in you build setup.

-Kay

shroffk, 2016-01-20:
The missing diirt plugins are due to the currently broken diirt publish job (this job is responsible for publishing the diirt 4.2 p2 repo)

The error has been escalated by cloudbees but still hasn't been resolved.
Once published diirt dependencies should be resolved.
I am working on this today and will update once this is done.

shroffk, 2016-01-22:
The diirt 4.2. p2 repository is ready
http://diirt.download.controlsystemstudio.org/diirt/3.0.2

@berryma4 can we include this in the 4.2 composite repo

berryma4, 2016-01-26:
I've updated the poms to the new version

berryma4, 2016-01-26:
These changes have also been cherry picked to 4.2.x

berryma4, 2016-01-26:
@CTschentscher, is everything working for you now?

CTschentscher, 2016-01-26:
Hi Eric,
I'm sorry, but it don't working  for me. 
I use nearly the same script like described by Kay, but make org.csstudio.product.
I'm not sure if the order of building is important, because in my script diirt is build after 
maven-osgi-bundles and cs-studio-tgirdparty. Theese 3 builds are success.

It failed in  cs-studio.core with nearly the same error like above:
[DEBUG] Registered artifact repository org.eclipse.tycho.repository.registry.facade.RepositoryBlackboardKey(uri=file:/resolution-context-artifacts@/Users/carla/Documents/CSS42/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui)
[DEBUG] tycho.localArtifacts=ignore -> ignoring locally built artifacts
[INFO] Resolving dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT @ /Users/carla/Documents/CSS42/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/pom.xml
[DEBUG] No solution found because the problem is unsatisfiable.: [Unable to satisfy dependency from org.csstudio.utility.pvmanager.ui 3.2.0.qualifier to package org.diirt.vtype 3.0.0.; No solution found because the problem is unsatisfiable.]
[INFO] {osgi.os=linux, osgi.ws=gtk, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier
[ERROR]   Missing requirement: org.csstudio.utility.pvmanager.ui 3.2.0.qualifier requires 'package org.diirt.vtype 3.0.0' but it could not be found
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT @ /Users/carla/Documents/CSS42/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/pom.xml: See log for details -> [Help 1]
org.apache.maven.MavenExecutionException: Cannot resolve dependencies of MavenProject: org.csstudio:org.csstudio.utility.pvmanager.ui:3.2.0-SNAPSHOT @ /Users/carla/Documents/CSS42/cs-studio/core/utility/utility-plugins/org.csstudio.utility.pvmanager.ui/pom.xml
	at org.eclipse.tycho.core.maven.TychoMavenLifecycleParticipant.afterProjectsRead(TychoMavenLifecycleParticipant.java:100)
	at org.apache.maven.DefaultMaven.doExecute(DefaultMaven.java:266)

If you are interested I can send you the whole Log-File.

The 4.1 branch is the last, that is success. But if I run the cs-studio.product and try to update to 4.2
it failed.

Best Carla

kasemir, 2016-01-26:
I'd build diirt first, but not sure if that's the problem in your case.

Your error is basically "Unable to satisfy dependency from org.csstudio.utility.pvmanager.ui 3.2.0.qualifier to package org.diirt.vtype 3.0.0", i.e. org.diirt.vtype is not found.

In my example on https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/build, I invoke maven with settings.xml which contain `<csstudio.composite.repo>${env.SRC_DIR}/org.csstudio.sns/css_repo</csstudio.composite.repo>`.
That composite repo, https://github.com/ControlSystemStudio/org.csstudio.sns/tree/master/css_repo, lists `<child location="../../diirt/p2/target/repository/"/>`, which has been created when building diirt.
Inside that  diirt/p2/target/repository there's plugins/org.diirt.vtype*.jar, so when building cs-studio/core, it's found.

Maybe follow those breadcrumbs:
Do you have  diirt/p2/target/repository/plugins/org.diirt.vtype*.jar?
Is diirt/p2/target/repository included in your composite repo?
Is that composite repo used by maven when building cs-studio/core?

CTschentscher, 2016-01-29:
Hi Kay,
ok I changed my settings.xml from:
`<csstudio.local.repo>$HOME/tmp/my_build_repo</csstudio.local.repo>`
to 
`<csstudio.composite.repo>$HOME/../CSSnightly/css_compo</csstudio.composite.repo>`
and create the files compositeArtifacts.xml and compositeContent.xml like listed in your css_repo.
I assumed, that during the build css_compo is filled with features  and plugins like using
csstudio.local.repo.

But if I build there is no repository in css_compo and the core build failed with this error:
`[ERROR] Internal error: java.lang.RuntimeException: Failed to load p2 repository with ID 'csstudio-composite-repo' from location file:/Users/carla/Documents/CSSnightly/css_compo: No repository found at file:/Users/carla/Documents/CSSnightly/css_compo. -> [Help 1]
`
the answer to your  breadcrumbs are:
Do you have diirt/p2/target/repository/plugins/org.diirt.vtype*.jar?  YES 
Is diirt/p2/target/repository included in your composite repo?  NO There is nothing in 
Is that composite repo used by maven when building cs-studio/core? YES see ERROR

Thanks for your help.
Carla

closed 2020-07-23 by shroffk