ControlSystemStudio/cs-studio #1571: diirt util ignores the osgi services

shroffk, 2016-01-20:
Some of the changes that were made to allow re configuring of the default composite datasources to using preferences results in loss of osgi configured providers.

shroffk, 2016-01-20:
I have added a fix, which only updates the configuration of the composite datasource while leaving the registered providers untouched.

berryma4, 2016-01-26:
It seems like the ~/.diirt directory and contents with defaults still gets created, although everything does use the directory configured in the preferences.ini correctly.
This might seem confusing though.

shroffk, 2016-01-26:
Well, this is there since osgi services and java services are registered before the workspace is created...which triggers the creation of the default config folder.

We are creating a contradictory(insane) set of requirements where we want to use design patterns that takes care of the loading/unloading of modules..since we don't want to bother with all that un interesting lifecycle stuff...but then we want fine grained control over the life cycle (something strongly discourage in both the java ds and osgi services) to inject preferences, which are acquired from the workbench.

closed 2016-01-26 by shroffk