ControlSystemStudio/cs-studio #1574: Saving perspective doesn't save contents of OPI views

willrogers, 2016-01-22:
* Open new workspace
* Create new OPI
* Open OPI in workbench
* Save perspective with new name "Test"
* Window | New Window

The new window will open in the Test perspective, but the OPI view will be empty.

I'm puzzling over this a bit.  When you shut down CSS, `saveState()` is called and you can save data to the memento.  Then when you re-open CSS the view is correctly restored.  However, if you save a perspective `saveState()` is not called, and when the perspective is reset no memento is provided to the view.

I don't know the best way to handle this.  @kasemir, do you know about this?

kasemir, 2016-01-22:
Looks to me like the perspectives are per window.

If I create a perspective in Window 1, I can close, reset, get it on restart.

But in Window 2, such a perspective won't work. The perspective shows up in the list of perspectives, but is deficient as you describe.

You can still create a perspective where you start in window 1, open a bunch of panels, drag some of them out into their own separate window (*), then save that as a perspective
-> Now you do have a saved perspective with multiple windows. As you switch perspectives, those separate windows will open/close depending on the active perspective.

.. But those separate windows (*) are 'detached' windows that belong to the same original workbench window. Only the original workbench has a menu and tool bar, the detached windows associated with it do not have menu nor toolbar.

So bottom line is that I can create several perspectives for a beam line, they can consist of one or more windows which open/close with the perspective, as long as I keep it to one main workbench window. If I use Window|New to open additional workbench windows, the perspectives don't quite work for the new window.

I agree that this is unfortunate, but not sure what to do about it. Also no show stopper, since I can create the various layouts as long as I create separate windows as detached windows, and not as new workbench windows. To users that's actually a bit better, because when we used multiple workbench windows, File/Exit in one would close all of CSS. With just one workbench window (and menu and toolbar), the other windows being detached windows (no menu, no toolbar), there is only one File/Exit, and there's no surprise that it indeed closes CSS.

willrogers, 2016-01-27:
I've looked at this for a while and it is a pain in the bottom.

If you ever open another window, then close your original window first, you can't get back to a working version of your saved perspective.  As you say, it is a perfectly valid workflow to use only one window, but right now the default workflow for editing OPIs introduces multiple windows, because pressing the "play" button in OPI editor launches the running OPI in a new window.

Also, this may present problems when we come to look at sharing perspectives if we want OPIs in them.

willrogers, 2016-02-02:
As an aside, if you get an empty OPIView (Window | Show View | BOY | OPI View) and right-click on it, exceptions are thrown.  The pull request above fixes that.

willrogers, 2016-04-01:
I've got another little bug.  After my perspectives fix, there's still one specific problem:

* _don't_ specify a location to save perspectives to
* right-click and Save As... on a perspective containing OPI files
* Window | New Window will open a new window in the new perspective, but the OPIs won't be restored

I've got a fix for this.

closed 2016-04-06 by willrogers