ControlSystemStudio/cs-studio #1588: ProcessVariable -> DataBrowser Context menu does not work

jbobnar, 2016-02-02:
When clicking on a widget (e.g. text input widget) and selecting DataBrowser from the popup menu, the databrowser does not open, or if it does the data are not fetched for the selected PV.

willrogers, 2016-02-02:
Can we get this fix on `4.2.x`?

berryma4, 2016-02-02:
I'll cherry pick to 4.2.x

jbobnar, 2016-02-04:
This has been merged and cherry-picked to 4.2.x as well.

closed 2016-02-04 by jbobnar

kasemir, 2016-02-05:
Thanks for identifying and fixing this!

Since the context menu had been OK for many years, any idea how the problem got introduced?
Is the issue only about the 'ChannelInfo' internal to the data browser, or do other handlers of PVName context menus also need to be updated?

jbobnar, 2016-02-05:
@kasemir 
It got broken by this commit dc7fd26b - getActiveMenuSelection was replaced by getCurrentSelection. I guess getCurrentSelection works OK for most structured viewers.

willrogers, 2016-02-05:
Yes, it must have been introduced by my commit, so sorry for that.  I think I tested only from the databrowser view, where the current selection was the same as the active menu selection.  Jaka's fix will also avoid `ClassCastException` in the apparently unlikely event that what is retrieved is an `ISelection` but not an `IStructuredSelection`.