ControlSystemStudio/cs-studio #1591: Inconsistency setting up macros

willrogers, 2016-02-02:
In `ExternalOpenDisplayAction.openDisplay()`, macros are parsed with the default setting `useParentMacros` as true.

In `OpenDisplayAction.getInputMacros()`, a `MacrosInput` object is created with `useParentMacros` as false.

Because of the way objects have defined `equals()`, this means we can open two OPIShells with the same file and macros, when we want them to be unique.

From what @kasemir has said about macros, I guess that `OpenDisplayAction.getInputMacros()` should default `useParentMacros` to true.

kasemir, 2016-02-05:
I agree with always setting `useParentMacros=true`. I'd even suggest removing that option and treat macro definitions like environment variables in a Unix shell, i.e. they're always inherited, where the child either uses them, replaces them with a new value, or ignores them. Does anybody see a use case for _not_ inheriting them other than trying to save memory for macros that the child doesn't know about, yet since they're inherited they are still present?

willrogers, 2016-02-08:
I remember that you figured out that this behaviour was unnecessary and it's not there in Display Builder.

Is it worth removing this from BOY?  I don't know how wide the changes would have to be.  I guess this fits in with the macros cleanup in #740.

willrogers, 2016-02-11:
I think you can close this ticket and we can continue discussing macros in the other tickets.

closed 2016-02-11 by berryma4