ControlSystemStudio/cs-studio #1594: Move ADL and EDM converters from opibuilder.feature to separate features

mfurseman, 2016-02-04:
At the moment the opibuilder feature contains the following plugins:
- org.csstudio.opibuilder.converter
- org.csstudio.opibuilder.adl2boy
- org.csstudio.utility.adlParser

As far as I know these are only useful to people who have existing EDM or MEDM files, I would like to move them to two separate features, for instance:
org.csstudio.applications.opibuilder.edm.feature
org.csstudio.applications.opibuilder.medm.feature

This would allow people who want these plugins to explicitly include the features in their product configurations.

willrogers, 2016-02-04:
+1 

In particular in our product we have the 'ADL Parser' and 'ADL-2-BOY' perspectives, and they are not useful to us.

kasemir, 2016-02-05:
We'd likely include both the EDM and MEDM support, but having them in optional, installable features instead of the primary opibuilder feature is of course fine.

berryma4, 2016-02-10:
@shroffk, @willrogers, @utzeln, @mjclarke01, @jpenning, @ksdj
Please take note, this merge may require changing your product file.

closed 2016-02-10 by berryma4