ControlSystemStudio/cs-studio #1596: OPIShell may not be correctly disposed

willrogers, 2016-02-08:
It is possible to get a situation where an OPIShell is not correctly disposed and subsequent OPIShells do not work correctly.  This happens when the display inside an OPI shell is changed and the OPIShell cache is not correctly updated.

I have a fix for this.

willrogers, 2016-02-09:
This is code that only I am familiar, so to help with reviewing:

* A cache of OPIShells are retained so that the OPIShellSummaryView can list them
* OPIShell equality is defined by opi-file + macros
* If you change the OPI file inside the OPIShell, its equality properties change
* This means that the new version of OPIShell is no longer in the cache
* When the shell is now closed, it can't be removed from the cache, but it is disposed.

The change removes the 'old' shell from the cache and adds the 'new' one when the OPI file is changed.

berryma4, 2016-02-10:
can we get this on master?

closed 2016-02-10 by berryma4